jerryscript-project/jerryscript #641: Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT' failed in ecma_get_object_from_value

renatahodovan, 2015-09-21:
###### Jerry version:
    Checked revision: 6697523
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
Object.freeze(RegExp.prototype.compile)();
```

###### Backtrace:
```
ICE: Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-value.cpp(ecma_get_object_from_value):329.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=9502, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000436073 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004360c2 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (
    assertion=0x4c42e0 "ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT", 
    file=0x4c4138 "jerryscript/jerry-core/ecma/base/ecma-helpers-value.cpp", 
    function=0x4b2910 <_ZZ26ecma_get_object_from_valuejE8__func__.lto_priv.629> "ecma_get_object_from_value", line=329)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:97
# 5  0x0000000000456bd9 in ecma_get_object_from_value (value=4) at jerryscript/jerry-core/ecma/base/ecma-helpers-value.cpp:329
# 6  0x000000000041c531 in ecma_builtin_regexp_prototype_compile (this_arg=4, pattern_arg=4, flags_arg=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp-prototype.cpp:65
# 7  0x000000000041c485 in ecma_builtin_regexp_prototype_dispatch_routine (builtin_routine_id=200, this_arg_value=4, arguments_list=0x0, arguments_number=0)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp-prototype.inc.h:90
# 8  0x000000000049b7dc in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_REGEXP_PROTOTYPE, builtin_routine_id=200, this_arg_value=4, 
    arguments_list=0x0, arguments_number=0) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:163
# 9  0x000000000049aec8 in ecma_builtin_dispatch_call (obj_p=0x8e2990 <mem_heap+1552>, this_arg_value=4, arg_collection_p=0x8e2920 <mem_heap+1440>)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:478
# 10 0x000000000044bc21 in ecma_op_function_call (func_obj_p=0x8e2990 <mem_heap+1552>, this_arg_value=4, arg_collection_p=0x8e2920 <mem_heap+1440>)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:892
# 11 0x0000000000433ba1 in opfunc_call_n (instr=..., frame_ctx_p=0x7fffffffcf50) at jerryscript/jerry-core/vm/opcodes.cpp:829
# 12 0x00000000004a5ae5 in vm_loop (frame_ctx_p=0x7fffffffcf50, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:545
# 13 0x00000000004a5d89 in vm_run_from_pos (header_p=0x8e2840 <mem_heap+1216>, start_pos=1, this_binding_value=523, lex_env_p=0x8e27a0 <mem_heap+1056>, 
    is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:637
# 14 0x00000000004a5805 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:415
# 15 0x00000000004abebe in jerry_run () at jerryscript/jerry-core/jerry.cpp:1464
# 16 0x00000000004aa878 in main (argc=3, argv=0x7fffffffd558) at jerryscript/main-linux.cpp:458
```

egavrin, 2015-11-19:
Same for master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

closed 2016-02-10 by LaszloLango