jerryscript-project/jerryscript #644: Unexpected token error when using for loop without block

rtakacs, 2015-09-23:
<b>Jerry version:</b>

```
  Checked revision: 6697523
```

<b>Test case:</b>

```js
function a()
{
  for (i = 1; i < 5; i++)
    if (i) {
      5;
    }
}

```

It returns with <b>'ERROR: Ln 8, Col 1: Unexpected token'</b> however it is a valid JavaScript code.

<b> Expected behaviour: </b>
```
For loop should be parsed correctly without needing a block.
```

rtakacs, 2015-10-15:
The following variable declaration is wrong as well:
```javascript
aaa=function(){for(;;) return "I am aaa"},bbb=function(){return "I am bbb"};
```
Error message: <b>ERROR: Ln 1, Col 42: Unknown token ,</b>

Note: If I put the instructions of the for statement into a block, everything is ok.

rtakacs, 2015-10-15:
I've found an other bug:

```javascript
{for(;;)try{}catch(s){}}
```
Error message: <b>ERROR: Ln 1, Col 25: Unknown token End of file</b>

Note: if the instructions are not in block, everything is ok.

LaszloLango, 2016-02-25:
Works on current master (76b37f3)

closed 2016-02-25 by LaszloLango