jerryscript-project/jerryscript #645: Supporting discontinued segment of RAM for Heap

seanshpark, 2015-09-24:
For some embed device with discontinued RAM area it would be good to set Jerry Heap to be configured as several segments, say 2 or 4, where currently we can set only one.

For example, some STM32F4 has two address of RAM, one is normal RAM from 0x20000000 size 128K and special RAM called CCM from 0x10000000 size 64K. If we can set JerryHeap to be configured to reside in two ares, we can make JerryHeap more bigger. Though we can manually configure loader script to set specific static variables to CCM but it can have other issues.

egavrin, 2015-09-24:
In case of STM32, linker script can be configured to use CCM and normal RAM as single segment, so we'll have 192k of RAM.

Usually, CCM is completely reserved by OS for its own needs, is it true in our case?

seanshpark, 2015-09-24:
Case of NuttX, using CCM is optional and can be added to normal heap area, for malloc and free. The problem I encountered is that CCM has no connection with `instruction` so, but not sure as I didn't investigated yet, structures with `callbacks` may not correctly work if they are located in CCM section or it may be related with DMA, as CCM cannot work with DMA. I've tested with moving `net` related buffers in nuttx to CCM and it failed to work.

As for jerry, we can map alternative heap by assigning it's section like this.

1) add new section `.ccmram` in `ld.script` file.

```
MEMORY
{
    flash (rx) : ORIGIN = 0x08000000, LENGTH = 1024K
    sram (rwx) : ORIGIN = 0x20000000, LENGTH = 112K
    ccmram (rw) : ORIGIN = 0x10000000, LENGTH = 64K
}

OUTPUT_ARCH(arm)
ENTRY(_stext)
SECTIONS
{
...
	.ccmram (NOLOAD) : {
		. = ALIGN(4);
		_sccmram = ABSOLUTE(.);
		*(.ccmram)
		*(.ccmram*)
		. = ALIGN(4);
		_eccmram = ABSOLUTE(.);
	} > ccmram
...
```

2) assign alternate heap to this section
```
mem_heap_t mem_heap;
mem_heap_t __attribute__((section (".ccmram"))) mem_heap_alt;
```

3) and lots of codes for `mem_heap_alt`

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg