jerryscript-project/jerryscript #653: Conditional operator is not supported in case clauses

renatahodovan, 2015-10-02:
###### Jerry version:
    Checked revision: 98fc409
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var a = "foo";
switch(a) {
    case true ? "foo" : "bar":
        print("OK");
        break;
}
```
Output:
```
    case true ? "foo" : "bar":
                             ^
ERROR: Ln 3, Col 30: Expected either ';' or newline token

```

According to ecma5, any expression can stand in a case clause after the 'case' keyword but ternary statements are rejected in jerry.

wateret, 2015-10-30:
@renatahodovan 
I think we can close this. Check out #666.

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin