jerryscript-project/jerryscript #661: question about jerry benchmark and jerry snapshot

ZidongJiang, 2015-10-10:
Hi
Web page http://samsung.github.io/jerryscript/benchmark/benchmark.html shows the performance of jerryscripts. But I have some questions about the page.

1, What is jerry-snapshot and how to build it?
2, Which cases in sunspider and ubench are run during the performance test? How can we reproduce the whole test?

Thanks

lemmaa, 2015-10-11:
Hi @jiangzidong , Sorry to cause confusion. It was an accidental expose of our experimental benchmark page. We are currently tuning the benchmark environment to get the stable result. We will publish it soon with all benchmark scripts and detailed procedures.

Just for information, you can check the list of sunspider cases in comment of https://github.com/Samsung/jerryscript/pull/643#issuecomment-142269304.

For jerry-snapshot, it's experimental production feature of JerryScript to save runtime memory usage for JavaScript framework, such as IoT.js. It translates framework code which written in JavaScript to byte-code during JerryScript build, and link to binary image of JerryScript. Simply, it moves byte-code from RAM area to Flash area, so JerryScript can save the certain amount of heap space for storing byte-code.

Thanks

jiangzidong, 2015-10-12:
Got it! Thanks @lemmaa

weilianglin, 2015-10-12:
@lemmaa, Does the snapshot increase the binary size of JerryScript/IoT.js? Since you need to link both JavaScript code and bytecode.

lemmaa, 2015-10-12:
@weilianglin, In case of using snapshot in IoT.js, currently we store only bytecode for native modules instead of JavaScript code, and size of bytecode is pretty similar to JavaScript code. So not much increase in the binary size. Anyway, still it's an experimental result. :)

closed 2016-02-11 by LaszloLango