jerryscript-project/jerryscript #662: how to set "MEM_STATS" build configuration

ZidongJiang, 2015-10-10:
HI
I build the "jerry" with the steps in http://samsung.github.io/jerryscript/how-to/
```
make release.linux
```

But when I type 
```
./jerry --mem-stats xxx.js
```
it shows
```
Ignoring memory statistics option because of '!MEM_STATS' build configuration.
```

1, How can I set the MEM_STATS when building?
2, Is there other flags ? I didn't find then in makefile?

Thanks

ruben-ayrapetyan, 2015-10-10:
Hello, @jiangzidong.

To build engine with memory statistics, please, use `release.linux-mem_stats` target:
```sh
make release.linux-mem_stats
./build/bin/release.linux-mem_stats/jerry --mem-stats script.js
```

jiangzidong, 2015-10-12:
@ruben-ayrapetyan Got it, Thanks!
And there are other modes in Makefile, such as cp cp_minimal mem_stress_test
Maybe we should describe what's their main feature and purpose, as comments in Makefile or content in readme/wiki :)

closed 2015-11-30 by egavrin