jerryscript-project/jerryscript #664: Enabling setting MEM_HEAP_AREA_SIZE for linux host.

ILyoan, 2015-10-13:
On linux host machine, `CONFIG_MEM_HEAP_AREA_SIZE` would be set 256KB by default. I cannot find the way to set the configuration from CMake build system.

I would be useful to set the heap size for host machine so that we can test with memory consumption problem on host without flashing binary into target device.

egavrin, 2015-10-13:
It is possible to specify heap size in this file: https://github.com/Samsung/jerryscript/blob/master/jerry-core/config.h#L45

ILyoan, 2015-10-16:
@egavrin Yes, sure we can. but I think it would be better to set heap limit as build parameter, without touching source file.

akosthekiss, 2016-08-29:
The new build system now allows passing `-DMEM_HEAP_SIZE_KB=...` to cmake, which will define the `CONFIG_MEM_HEAP_AREA_SIZE` macro for the compiler.

closed 2016-08-29 by akosthekiss