jerryscript-project/jerryscript #665: Property lookup cache prevents GC

ILyoan, 2015-10-13:
It looks like from `ecma_lcache_insert()`, adding a property of an object increases RC for the object. This sometimes could prevent GC. 

I'm not sure I'm right about it, but I've reached tentative conclusion by exploring this for a while that this could be significant . For example, if we have a function only called once at the begining of program.
```
function foo(a) {
  ...
}
```
A lexical environment object is created for executing the function, and after finishes, the lexical object could not be freed by GC since a property of it is cached.

Please note that as my investigation is incomplete, the truth could be different. But I think we need to check this.

ruben-ayrapetyan, 2015-10-26:
@ILyoan, lcache actually could prevent GC, causing false positive out-of-memory.

One of possible solutions is to consider lcache's references as weak references in GC.

zherczeg, 2016-07-28:
Lcache has only weak references now.

closed 2016-07-28 by zherczeg