jerryscript-project/jerryscript #680: ICE: Assertion 'lit_ids != NULL' failed in var_to_str

ILyoan, 2015-10-22:
###### Jerry version:
Checked revision: 08c618e8
Build: debug.linux  

###### Test case:
```
f1(function(a1){});
```

run with --show-opcodes

ILyoan, 2015-10-22:
ICE occurs while printing VARG meta_op of `f1` function after prints function expression.

Because `vargs_num` and `seen_vargs` are not stacked, the values of  it will be corrupted after treating the function expression as first parameter of `f1`.

wateret, 2015-12-02:
Even though there have been no commits related to the issue, @ILyoan's test case somewhat no longer emits an error.

I got another test case that emits the same error, probably the same reason.
```javascript
var x = {
  foo: function () { throw new TypeError("bar"); }
};
```

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango