jerryscript-project/jerryscript #681: Jerry API for retrieving heap stats.

ILyoan, 2015-10-22:
For profiling and tracing memory usage during running a program. It would be useful to have API for getting heap memory stats.

Interesting numbers are total heap size, allocated bytes, and peak bytes.

```
typedef struct jerry_api_heap_status_t 
{
  size_t heap_size;
  size_t allocated_bytes;
  size_t peak_allocated_bytes;
} jerry_api_heap_stats_t;

void jerry_api_heap_stats (jerry_api_heap_stats_t*);
```

egavrin, 2015-10-22:
Important numbers are:
```   Peak allocated chunks count = 1386
   Peak allocated = 33469 bytes
   Peak waste = 55235 bytes
```

Currently available:
```
Heap stats:
   Heap size = 258048 bytes
   Chunk size = 64 bytes
   Allocated chunks count = 0
   Allocated = 0 bytes
   Waste = 0 bytes
   Peak allocated chunks count = 1386
   Peak allocated = 33469 bytes
   Peak waste = 55235 bytes

Pools stats:
  Chunk size: 8
   Pools: 0
   Allocated chunks: 0
   Free chunks: 0
   Peak pools: 380
   Peak allocated chunks: 3039
```

egavrin, 2015-10-22:
Additionally, it'd be nice to have API for resetting statistics: `void jerry_api_mem_stats_reset ();` - since we don't need to know the allocated peek after parsing, for example.

ILyoan, 2015-11-04:
@egavrin Thanks for your opinion, I agree with you. I'll prepare PR for this issue.

zherczeg, 2017-01-31:
No plans to implement this at the moment.

closed 2017-01-31 by zherczeg