jerryscript-project/jerryscript #682: why don't use object pointer at argument for "this" on external function callback?

idkiller, 2015-10-22:
```c++
typedef bool (*jerry_external_handler_t) (const jerry_api_object_t *function_obj_p,
                                          const jerry_api_value_t *this_p,      
                                          jerry_api_value_t *ret_val_p,         
                                          const jerry_api_value_t args_p[],     
                                          const jerry_api_length_t args_count); 
```

This is current callback type for external function.
I wonder why this_p is pointer of value.
I think "this" is only useful when function is constructor.
And also following code is part of current unit test for constructor.

```c++
static bool                                                                        
handler_construct (const jerry_api_object_t *function_obj_p,                       
                   const jerry_api_value_t *this_p,                                
                   jerry_api_value_t *ret_val_p,                                   
                   const jerry_api_value_t args_p[],                               
                   const jerry_api_length_t args_cnt)                              
{                                                                                  
  printf ("ok construct %p %p %p %d %p\n", function_obj_p, this_p, args_p, args_cnt, ret_val_p);
                                                                                   
  JERRY_ASSERT (this_p != NULL);                                                   
  JERRY_ASSERT (this_p->type == JERRY_API_DATA_TYPE_OBJECT);                       
                                                                                   
  JERRY_ASSERT (args_cnt == 1);                                                    
  JERRY_ASSERT (args_p[0].type == JERRY_API_DATA_TYPE_BOOLEAN);                    
  JERRY_ASSERT (args_p[0].v_bool == true);                                         
                                                                                   
  jerry_api_set_object_field_value (this_p->v_object, (jerry_api_char_t *) "value_field", &args_p[0]);
                                                                                   
  jerry_api_set_object_native_handle (this_p->v_object,                            
                                      (uintptr_t) 0x0012345678abcdefull,           
                                      handler_construct_freecb);                   
                                                                                   
  return true;                                                                     
} /* handler_construct */                                                          

```

`JERRY_ASSERT (this_p->type == JERRY_API_DATA_TYPE_OBJECT);`
It show "this_p" must be object.

ruben-ayrapetyan, 2015-11-02:
@idkiller,

In the following example, value of `x` would be number, equal to 1, not object.
```js
Number.prototype.test = function () { 'use strict'; return this; }
var x = (1).test()
```

So, js supports non-object `this` values. However, I'm also unsure if it can be really useful.

<br/>
> I think "this" is only useful when function is constructor.

Also `this` is useful in any other functions, if the functions are bound to some object. In this way, js emulates OOP.

idkiller, 2015-11-04:
@ruben-ayrapetyan 
Thank you! I understand it.

closed 2015-11-04 by idkiller