jerryscript-project/jerryscript #683: mem_free_chunks_number is not zero on clean up after "jerry_api_set_object_native_handle" is called twice

idkiller, 2015-10-22:
`jerry_cleanup()` always make assert error after some object native handle were updated.

```c++
#include "jerry.h"
const char *jscode = "foobar = {};";
jerry_init(JERRY_FLAG_EMPTY);
jerry_parse (reinterpret_cast<const jerry_api_char_t*>(jscode), strlen (jscode));
jerry_run();
jerry_api_object_t *g = jerry_api_get_global();
jerry_api_value_t foobar;
jerry_api_get_object_field_value(g, reinterpret_cast<const jerry_api_char_t*>("foobar"), &foobar);
jerry_api_set_object_native_handle(foobar.v_object, static_cast<uintptr_t>(0xabcdef), nullptr);
jerry_api_set_object_native_handle(foobar.v_object, static_cast<uintptr_t>(0x123456789), nullptr);
jerry_api_release_object(g);
jerry_api_release_value(&foobar);
jerry_cleanup();
```

this code always failed.
```
ICE: Assertion 'mem_free_chunks_number == 0' failed at jerryscript/jerry-core/mem/mem-poolman.cpp(mem_pools_finalize):183.
Error: ERR_FAILED_INTERNAL_ASSERTION

```

Is this bug? or intended design?
comment on function note shows that can be updated.
```
If native handle was already set for the object, its value is updated.
```

egavrin, 2015-10-22:
`mem_pools_finalize` checks if all allocated memory is freed upon exit to prevent memory leaks. This assert means that some of allocated data wasn't released.

idkiller, 2015-10-22:
yes, I knew about `mem_pools_finalize`, but in my code, I release every created object and value.
I guess allocated memory would be missed when `jerry_api_set_object_native_handle` is called.
I wonder `jerry_api_set_object_native_handle` is designed to call only one time as one object.

ruben-ayrapetyan, 2015-10-26:
@idkiller, yes, seems that there is a bug: `ecma_create_external_pointer_property` should not reallocate "external pointer" (`ecma_external_pointer_t *handler_p = ecma_alloc_external_pointer ();`) if it was already allocated.

closed 2015-11-10 by idkiller