jerryscript-project/jerryscript #686: argument and eval in strict mode

zherczeg, 2015-10-23:
I am working on the strict mode parsing and Jerry seems accepting the following at the moment:

```
function f(eval) { "use strict" }
```

The spec says this is not allowed:
http://www.ecma-international.org/ecma-262/5.1/#sec-13.1

However, Firefox also seems to reject the following, and I am not sure this is correct:

```
function eval() { "use strict" }
```

Since eval is created in the non-strict world, and I haven't found anything that prevents this (so far), I think we should accept this. What do you think?

ruben-ayrapetyan, 2015-10-25:
@zherczeg, seems that specification disallows the second case also, as `Identifier` is name of declared function (http://www.ecma-international.org/ecma-262/5.1/#sec-13):
```
It is a SyntaxError if the Identifier "eval" or the Identifier "arguments" occurs
as the Identifier of a strict mode FunctionDeclaration or FunctionExpression.
```

zherczeg, 2015-10-26:
You are right, I wasn't observant enough.

ruben-ayrapetyan, 2015-10-30:
Fixed

closed 2015-10-30 by ruben-ayrapetyan