jerryscript-project/jerryscript #691: Date built-in (test262 failures)

ruben-ayrapetyan, 2015-10-29:
Hello.

The following test262 tests related to `Date` built-in are failing on current master (906558b8e94aa9118971188af9880eb47ff9b791):
- ch15/15.9/15.9.3/S15.9.3.1_A5_T1
- ch15/15.9/15.9.3/S15.9.3.1_A5_T2
- ch15/15.9/15.9.3/S15.9.3.1_A5_T3
- ch15/15.9/15.9.3/S15.9.3.1_A5_T4
- ch15/15.9/15.9.3/S15.9.3.1_A5_T5
- ch15/15.9/15.9.3/S15.9.3.1_A5_T6

galpeter, 2015-10-29:
This could be a related to https://github.com/Samsung/jerryscript/issues/213

dbatyai, 2015-10-30:
The problem is that the tests were created for a specific timezone (namely US/Pacific) and fail when they run in anything other. Also, as far as I know there's no functionality in jerry-libc for getting system timezone, so in our case these always run with GMT time.

ruben-ayrapetyan, 2015-10-30:
@dbatyai, thank you for checking.

closed 2016-02-20 by LaszloLango