jerryscript-project/jerryscript #699: Nasty (very difficult to fix) parser scanner bug

zherczeg, 2015-11-02:
for (/;/.x in b) ;
           ^
ERROR: Ln 1, Col 12: Expected ';' token

This is a perfectly valid statement. The problem is that a slash can be a start of a regex pattern (primary expression), or a binary divide operator. Without deep analysis it is virtually impossible to tell the role of the slash (primary expression or binary operator). The worst thing is that the characters after the slash are part of a regex if the slash marks the start of a regex.

I run the same problem in the compact byte code parser, and I don't see any trivial fix for this.

LaszloLango, 2016-02-09:
Fails on current master (abc95d3)

zherczeg, 2016-02-09:
Fails? Of course it fails since b is not defined. (But there should not be a parse error). This one prints a hello:

```
for (/;/.x in {}) ;
print("Hello");
```

LaszloLango, 2016-02-10:
@zherczeg, you are right. I missed that and thought it is still a parse error, but it's not. I close the issue.

closed 2016-02-10 by LaszloLango