jerryscript-project/jerryscript #702: Logo creation

Seo-Young, 2015-11-04:
Our project has not the official logo yet. Thus, I would like to suggest to open an issue for logo creation.
Please post ideas into this thread. If you have any logo images to suggest, feel free post them. 
You do not need to be a designer. If you can explain your idea, community can make it real image.
Every suggested ideas will be our logo candidates. 
Logo iterations also welcome.

I would like to open this issue for three weeks and gathering enough candidates. 
This issue will be closed at 11/25.
I will open a new issue for voting with candidates 11/26. 
And I guess we will meet the first day of December with our new logo. :)

seanshpark, 2015-11-04:
Can we talk with MGM to use Jerry image?
Mix with JavaScript image found in the internet, in a yellow rounded rectangle with black silhouette of Jerry.

Seo-Young, 2015-11-05:
[image]

Is this the right image in your mind?
The distribution right had been moved to Warner Bros. since 1988. Do we have to ask them? Hum.. let's see.

seanshpark, 2015-11-05:
YES! :clap:  also there can be also with whole body

Seo-Young, 2015-11-05:
Some iterations
[image]

Seo-Young, 2015-11-05:
[image]

[image]

akiss77, 2015-11-05:
Some hand drawings. Certainly not full-fledged ideas.
[image]

ILyoan, 2015-11-05:
[image]

ILyoan, 2015-11-05:
[image]

IoT.js logo is bonus :)

zherczeg, 2015-11-05:
The last picture is quite similar to Mickey Mouse and I am sure Disney will not allow us to use it.

Seo-Young, 2015-11-05:
Let's ask to Disney! :)

ILyoan, 2015-11-06:
By the way, I would be unhappy if I was Disney and someone is asking me to allow using Mickey logo while call it Jerry :(

ILyoan, 2015-11-06:
[image]

Seo-Young, 2015-11-10:
I've checked the JS logo. It is released under a MIT license so we can use it in our logo as we want. 
Refer to https://github.com/voodootikigod/logo.js

Seo-Young, 2015-11-10:
[image]

[image]

@akiss77 's hand drawings. I missed the tail of the latter one.

idkiller, 2015-11-10:
[image]

Drawing with Inkscape, Just "J" cheese with small rat.

akanatov, 2015-11-11:
@idkiller if you put J-cheese under Jerry's right paw and S as $ (dollar) :-) under the left one - then it will be Java$cript :-)

tczene, 2015-11-11:
[image]

davidsz, 2015-11-20:
[image]

Some variations.

closed 2015-11-26 by Seo-Young