jerryscript-project/jerryscript #707: what's the purpose of vm_stack_push_value() in vm-stack.cpp

jiangzidong, 2015-11-06:
the vm_stack_push_value and vm_stack_pop_value deal with the dynamic variables in stack. Those vars will be stored in the inline array or allocated heap (top_chunk_p). Those stack variables also are "root" live object in gc.

But no one call vm_stack_push_value  in jerry. what's the purpose of the stack value? 

```
/**
 * ECMA stack frame
 */
typedef struct vm_stack_frame_t
{
  struct vm_stack_frame_t *prev_frame_p; /**< previous frame */
  vm_stack_chunk_header_t *top_chunk_p; /**< the top-most chunk of the frame */
  ecma_value_t *dynamically_allocated_value_slots_p; /**< pointer to dynamically allocated value slots
                                                      *   in the top-most chunk */
  uint32_t current_slot_index; /**< index of first free slot in the top chunk */
  ecma_value_t inlined_values[VM_STACK_FRAME_INLINED_VALUES_NUMBER]; /**< place for values inlined into stack frame
                                                                      *   (instead of being placed on heap) */
  ecma_value_t *regs_p; /**< register variables */
  int32_t regs_number; /**< number of register variables */
} vm_stack_frame_t;
```

jiangzidong, 2015-11-20:
is there anyone can help me understand this function?

ruben-ayrapetyan, 2015-11-20:
@jiangzidong, the functions are not currently used, but maybe would be used in future. If would not, then the functions would be removed.

closed 2015-12-01 by egavrin