jerryscript-project/jerryscript #710: JS Debugging with Eclipse

haosu1987, 2015-11-08:
I've used some other JS interpreter, which includes its own debugging protocol, and can be integrated into the IDEs. For example, Eclipse can debug both the Node.js and the JS in the Web with the ChromeDevTools Plugin that impliment by the [V8 debugger protocol](https://chromium.googlesource.com/v8/v8/+/master/docs/debugger_protocol.md) .
I think that embedded systems also need such a protocol with its associated debugging tools , to help us debug and develop JS projects.

lemmaa, 2015-11-09:
Hi, @haosu1987. Great! We also considering about that. Do you think it is suitable to support V8 debugger protocol in JerryScript? Or, do you have any idea better than this? As for IDE, is it a good choice integrate to Eclipse? How about Sublime Text, ATOM, Microsoft Code etc.?

haosu1987, 2015-11-09:
Hi, @lemmaa . Thanks for your prompt.

I think __V8 Debugger Protocol__ may not be suitable for the embedded system(especially for constrained devices). It's a large overhead that based on JSON format strings. Thus, we need a tiny protocol at first. It can be designed by a serial of compressed instructions (for example [Duktape Debug stream format](https://github.com/svaarala/duktape/blob/master/doc/debugger.rst#debug-stream-format)), that contains some basic functions, as:

* Informations include local variables, call stack and operating status in current environment
* That pause and run the program, implements the "pause", "resume", "step over", "step into" and "step out" which we usually use to debugging program
* Breakpoints and "debugger" statement support
* Get and put local variables in the current stack

Then, we can consider that how to debug the devices upon the protocol.

I choose the Eclipse because of the Chrome DevTools plugin. It's also an open source project that implements V8 Debugger Protocol. We can translate our protocol to V8 Debugger Protocol by a debug bridge, and may debug with Eclipse Chrome DevTools directly.

As for other IDEs, I think we can also support by the debug bridge. But, it may spend a little time to adapt.

egavrin, 2015-11-12:
@haosu1987 in case we're going to debug on hosts I'd prefer to have support of V8 Debugger Protocol, since it's widely adopted by IDEs and editors, like Eclipse, Atom etc. So, we won't spend much time on integration.

egavrin, 2015-11-12:
Anyway, we can start implementation of debugging capabilities in our shell #712, which would be the most difficult part of the task. Engine itself shouldn't have to support V8's, Duktape's or any other protocol - it should only provide API for debugging. Actual integration would be handled by debugger as @haosu1987 mentioned.

lemmaa, 2015-11-12:
@egavrin After all, we should implement debugger server as separate module, not in our shell, I think. Because it should be activated in embedded situation, such as IoT.js. 

As for features, it needs to support both serial connection and TCP/IP connection for remote debugging. We have to make it as light as possible, because it will run on the target in limited resource.

Therefore, IMHO, define protocol as light as possible is important in this task. Integration with IDEs, which support well known protocol such as V8's, can be done by bridge server (or module) which run on host PC.

weilianglin, 2016-06-28:
Ping~

Any progress on the JS debugger feature request? 

Thanks
-Weiliang

polaroi8d, 2016-06-28:
Me and @knightburton are working on this topic now

- At the moment we are able to insert breakpoints instructions before each statement (the script must run in debug mode)
- We have started to created a debugger tool which will be able to communicate with JerryScript through TCP/IP communication
- We are creating data structures for this tool to store the current status of JerryScript (living functions, breakpoints, register names)
- Currently we are updating the ports because the API of JerryScript has been changed recently

sergioamr, 2016-07-08:
@polaroi8d  https://github.com/knightburton/jerryscript/commits/remote-debugger
I am working in a web IDE to write and upload into the Arduino 101 using the serial port & later bluetooth. Still need to get approval to open source it but i think it would be interesting if i could add debugging functionality.

martijnthe, 2016-08-26:
@polaroi8d and @knightburton: this sounds interesting. Can you give an update on how far along you are with this? Later down the road, it would be interesting to be able to hook this up to the Blink Developer Tools (maybe by modifying https://github.com/node-inspector/node-inspector).

polaroi8d, 2016-08-29:
Sorry for the late answer @martijnthe. Now I am updating the repository, because there is a new build system. 

- The TCP/IP communication is ready to use
- There is a function name sender, which can send the parsed function names over the TCP/IP
- Next step is be able to send the function memory address when come into existence.

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg