jerryscript-project/jerryscript #711: jerry-test-suite produces strange result

chunseoklee, 2015-11-10:
```sh ./tools/runners/run-test-suite-jerry.sh  ./build/bin/debug.linux/jerry```
There are 15 failed test for test above.
But, some tests(15.5.2.1-001.js,15.5.4.7.00x.js ) seems to work well in my own test : make separate test.js file.  
I can not figure out the reason.

seanshpark, 2015-11-10:
```assert(true);
```
is ok.
```
  assert(1);
```
fails with `Script assertion failed`

galpeter, 2015-11-10:
The `assert(1)` fails because the assert's implementation currently only checks that the argument is a true boolean or not. (The implementation is in the main-linux.cpp file). I can create a PR which adds a case that if the argument is an integer 1 then it accept just like the boolean true.

akiss77, 2015-11-10:
Shouldn't any non-0 integer count as true, not only integer 1?

galpeter, 2015-11-10:
@akiss77, sure we can do that also, and maybe it's even better.

galpeter, 2015-11-10:
@akiss77 , should 0.01 be also a valid true? :)

akiss77, 2015-11-10:
```
>>> Boolean(0)
false
>>> Boolean(1)
true
>>> Boolean(2)
true
>>> Boolean(0.0)
false
>>> Boolean(1.0)
true
>>> Boolean(0.01)
true
>>> Boolean(NaN) 
false
```

LaszloLango, 2015-11-10:
IMHO not only integers, but every number should be true except zero.

akiss77, 2015-11-10:
What's more:
```
>>> Boolean("")
false
>>> Boolean("non-empty string")
true
>>> Boolean({})
true
```
I'd try to cast everything to boolean. Whatever yields true, should be accepted.

galpeter, 2015-11-10:
hmmm then the best way would be to expose an API call which can correctly convert the values to bool.

lemmaa, 2015-11-10:
@LaszloLango @akiss77 @galpeter, I'm currently working on it. I'm trying to define new API `jerry_apt_value_to_char_buffer()` to get character representation of `jerry_api_value_t` object. I conclude that it is best way to solve this issue after short investigation. I think it can be used to REPL implementation of @wateret. If you agree, I will keep going.

````
ssize_t
jerry_api_value_to_char_buffer (const jerry_api_value_t *value_p, /**< value descriptor */
                                 jerry_api_char_t *buffer_p, /**< output characters buffer */
                                 ssize_t buffer_size) /**< size of output buffer */

````

akiss77, 2015-11-10:
@lemmaa Why converting it to characters? First of all, I'm not sure that a char buffer representation will help us deciding whether it's a true-compatible value (but I may be wrong here), secondly, it causes an allocation of an (arbitrarily long?) buffer while we are only looking for a single boolean value.

lemmaa, 2015-11-10:
@akiss77 Yes, I agree that it's inefficient, only for this issue. :)
For current issue, we just need an API such as `jerry_api_is_value_true()`, to check the boolean result of value. But, IMHO, such an API is useless instead of solving this issue. So I'm considering more general purpose one. And I think that we don't need such a long buffer allocation, because we need only check the existence of well known strings. For example, `true`, `false`, `null`, `undefined`, 'NaN`, `0`, `0.0`, and `<OBJECT>` for object. Buffer overflow can handled inside Jerry.

If you have other idea for well suited API sets. I'm always welcome!

ruben-ayrapetyan, 2015-11-10:
If `assert` accepts any value that could be converted to boolean true, some issues can become hidden, like in the following test:
https://github.com/Samsung/jerryscript/pull/498/files#diff-c7d78eb307700cc443dc3324384a4033

```js
assert ("abc".replace("abc", "$x$$5$0$00$" === "$x$5$0$00$"));
```

The test returned some string that, if not empty, converts to boolean true.
However, the test is incorrect - right parenthesis is in a wrong place.
`assert` that implicitly converts values to boolean would accept any value except empty string, so the issue become hidden.

Actually, seems that the test should be expressed like the following (passing explicit boolean value to `assert`):
```js
assert ("abc".replace("abc", "$x$$5$0$00$") === "$x$5$0$00$");
```

Until `assert` was changed to accept only exact `true` value, the issue was hidden.

<br/>
So, in my opinion, its better to update tests so that only explicit boolean values are passed to `assert` call instead of updating implementation of `assert`.

In the #498 pull request only the tests that are currently enabled, were updated. We could update other tests as they become enabled.

zherczeg, 2015-11-10:
I would prefer to use assert (true) than assert (1), or introduce an assert_unreachable();

galpeter, 2015-11-10:
@lemmaa, @akiss77 what I had in mind is a general function which can covert any api value to boolean api value. Like jerry_api_to_boolean(..) that's a more general way imho and probably it will useful for others in the future.

galpeter, 2015-11-10:
@lemmaa @akiss77: this is my proposed API which should convert the value to boolean.

galpeter, 2015-11-10:
ahh sorry I've missed the link, so this is the API call I was thinking of: https://github.com/galpeter/jerryscript/commit/eebd4c83a03e78ecfbcaf6c0862be9b55c8d4e7c

akiss77, 2015-11-10:
@zherczeg Just for the sake of correctness, you must have meant `assert(false)` (and `assert(0)`) as an alternative to `assert_unreachable()`, right? (Since `assert(true)` should be no-op, AFAIK.) Personally, I'd still go with `assert(false)` instead of introducing yet another built-in to the jerry command line executable, but I have no strong opinions.

@ruben-ayrapetyan As for the semantics of `assert()`, I have no strong opinions either. It can be left as-is (i.e., accepting only boolean true and bailing out for anything else) or changed. BUT IF it is changed, then I'd certainly vote for using "to-boolean" semantics, i.e., converting any value to boolean according to JS rules and bailing out on false. All my previous comments were along this line, since the discussion was winding in the direction of accepting `assert(1)` as equivalent to `assert(true)`. Eventually, it's OK with me not going that path.

@lemmaa @galpeter I had something similar to galpeter's proposal in mind. It's clean, with low-overhead, and follows a naming scheme (`jerry_api_value_to_`) that may be useful for other types in the future.

lemmaa, 2015-11-10:
@ruben-ayrapetyan Basically, I agree with you. But at this time, I'd like to focus only this issue. 

@galpeter Good! What you did is what I want to be at first completely same way, before considering more general one. :)

@akiss77 I agree with you. How about define set of `jerry_api_value_to_` APIs at this time. Do you have any suggestion for this set, such like...
````
jerry_api_value_to_boolean
jerry_api_value_to_number
jerry_api_value_to_char_buffer
...
````

seanshpark, 2015-11-11:
:+1: to `jerry_api_value_to_` series.

ruben-ayrapetyan, 2015-11-11:
@lemmaa, yes, sure. I've raised issue #716.

zherczeg, 2016-07-28:
This is done.

closed 2016-07-28 by zherczeg