jerryscript-project/jerryscript #717: why does rss-measure.sh exclude the executable code's rss.

jiangzidong, 2015-11-12:
rss-measure.sh use ` grep -A 1 -- "r[w-]-p "` to exclude the `r-xp` part's rss, which is the executable part of jerry. And it's the main part which consumes memory. (about 280k)

So the result of rss-measue is different with other memory tool, such as smem.

Why does rss-measue.sh do that? 

```
jzd@jzd32:~$ cat /proc/11258/smaps
08048000-0808f000 r-xp 00000000 08:11 61344731   /home/jzd/jerryscript/j.l
Size:                284 kB
Rss:                 284 kB
Pss:                 284 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:       284 kB
Private_Dirty:         0 kB
Referenced:          284 kB
Anonymous:             0 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
VmFlags: rd ex mr mw me dw
0808f000-08091000 rw-p 00046000 08:11 61344731   /home/jzd/jerryscript/j.l
Size:                  8 kB
Rss:                   4 kB
Pss:                   4 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:         0 kB
Private_Dirty:         4 kB
Referenced:            4 kB
Anonymous:             4 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
VmFlags: rd wr mr mw me dw ac
08091000-082d2000 rw-p 00000000 00:00 0
Size:               2308 kB
Rss:                  20 kB
Pss:                  20 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:         0 kB
Private_Dirty:        20 kB
Referenced:           20 kB
Anonymous:            20 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
VmFlags: rd wr mr mw me ac
b7785000-b7786000 r-xp 00000000 00:00 0          [vdso]
Size:                  4 kB
Rss:                   0 kB
Pss:                   0 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:         0 kB
Private_Dirty:         0 kB
Referenced:            0 kB
Anonymous:             0 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
VmFlags: rd ex mr mw me de
b7786000-b7788000 r--p 00000000 00:00 0          [vvar]
Size:                  8 kB
Rss:                   0 kB
Pss:                   0 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:         0 kB
Private_Dirty:         0 kB
Referenced:            0 kB
Anonymous:             0 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
VmFlags: rd pf io de dd
bfb92000-bfbb3000 rw-p 00000000 00:00 0          [stack]
Size:                136 kB
Rss:                  12 kB
Pss:                  12 kB
Shared_Clean:          0 kB
Shared_Dirty:          0 kB
Private_Clean:         0 kB
Private_Dirty:        12 kB
Referenced:           12 kB
Anonymous:            12 kB
AnonHugePages:         0 kB
Swap:                  0 kB
KernelPageSize:        4 kB
MMUPageSize:           4 kB
Locked:                0 kB
```

egavrin, 2015-11-12:
@jiangzidong we're primarily targeting microcontrollers, where memory addressed directly and execution of application is performed from FLASH memory without loading executable into memory. So, for us is more important to understand how much of stack and heap are physically used by application. Measuring RSS is the most convenient way to that, since we're single-threaded and statically linked. This script is used as first-stage approximation to check if we're moving to the right direction and haven't introduced significant memory degradation.

To get more detailed memory information you can use `mem_stats` or `freya`:
```
./jerry.master --mem-stats ./t.js ; echo $?
Heap stats:
   Heap size = 258048 bytes
   Chunk size = 64 bytes
   Allocated chunks count = 0
   Allocated = 0 bytes
   Waste = 0 bytes
   Peak allocated chunks count = 636
   Peak allocated = 40632 bytes
   Peak waste = 205 bytes

Pools stats:
  Chunk size: 8
   Pools: 0
   Allocated chunks: 0
   Free chunks: 0
   Peak pools: 631
   Peak allocated chunks: 5041
```

jiangzidong, 2015-11-12:
@egavrin Thanks. 
And I have a question about the behavior of mem_heap in microcontrollers.

1. mem_heap is actually a global array. in PC, it is in .bss section and is not in the memory at first. When some chunk are visited, those chunks are moved into memory (RSS increased), and will never leave momery no matter if they are "deallocated". So the one purpose of memory optimization: **jerry try to avoid visiting new heap-chunk. ** a) lower the peak allcoated chunks. b)reuse the chunks which are visited before.  Am I right about that?

2. Is the microcontrollers' behavior of global array similar with PC? (only visited chunked will stored in RAM)

egavrin, 2015-11-12:
@jiangzidong in short, for MCU you usually have fixed amount of memory available and you cannot change this at runtime. Usually, OS reserves 16k of heap memory for you and app should operate within this limit.
So, returning to your questions:
1) Our first priority should be a) `lower the peak allcoated chunks` and b) `reuse the chunks which are visited before`. You're absolutely right - reduce peak and reduce fragmentation.
2) On MCU you don't have Virtual Memory -> all chunks of memory that were allocated directly in physical RAM. For example, in Linux you can dynamically allocate 100mb array in virtual space and Rss or Pss would less than 100mb until you touch them, this is called `Dirty`. In MCUs 1mb allocation means 1mb allocation.

jiangzidong, 2015-11-13:
@egavrin Got it. Thanks

So here is my understanding about 3 memory tools: rss-measure, mem-stats, and freya. please correct me if I'm wrong
* rss-measure: measure the size of jerry's stack vars, global vars, and touched global array (including "mem_heap")
* mem_stats: only measure the usage of "mem_heap", peak value matters.
* freya: similar with mem_stats, measure the usage of mem_heap. 
* non of those can reflect the memory fragmentation. rss-measure can show fragmentation slightly, because of the "Virtual Memory Dirty". But because the page in linux is 4K, the measurement is too coarse, and it cannot show the fragment among each 4K pages

egavrin, 2015-11-13:
You got it right :)
Regarding fragmentation, `mem-stats` prints amount of Waste and Waste Peak,
which are basically used to estimate fragmentation.   This would great
debugging feature for the engine to print detailed memory map showing the
usage per object/feature and see the root of fragmentation. Additionally,
we don't have any "commonly accepted" way to check stack usage yet.

closed 2015-11-16 by jiangzidong