jerryscript-project/jerryscript #718: make release.linux USE_COMPILER_DEFAULT_LIBC=YES fails

akiss77, 2015-11-12:
Lately, I've been trying to build Jerry on a platform which is not supported by jerry-libc (aarch64-linux), thus I had to fall back to the default glibc available there. Debug build worked fine, however, release build failed. Then it turned out that the error is also reproducible on an x86-64 box as well:

```
/usr/bin/x86_64-linux-gnu-g++    -ffixed-rbp    -ffixed-rbp -fno-builtin -flto -fno-fat-lto-objects -fno-stack-protector -g -gdwarf-4 -Wall -Wextra -pedantic -Wformat-nonliteral -Winit-self -Wno-stack-protector -Wconversion -Wsign-conversion -Wformat-security -Wmissing-declarations -Wno-attributes -Wfatal-errors -Werror -Wlogical-op -std=c++11 -fno-exceptions -fno-rtti -fno-implicit-templates -fno-implicit-inline-templates -Os -Wl,-z,noexecstack -flto  -static CMakeFiles/release.linux.dir/main-linux.cpp.o  -o release.linux  jerry-core/librelease.jerry-core.a third-party/fdlibm/librelease.jerry-fdlibm.third_party.lib.a /usr/lib/gcc/x86_64-linux-gnu/4.9/libgcc.a /usr/lib/gcc/x86_64-linux-gnu/4.9/libgcc_eh.a 
In function ‘__fread_alias’,
    inlined from ‘read_snapshot’ at /home/akiss/devel/jerry/jerryscript/main-linux.cpp:153:69,
    inlined from ‘main’ at /home/akiss/devel/jerry/jerryscript/main-linux.cpp:390:70:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:290:66: error: call to ‘__fread_chk_warn’ declared with attribute warning: fread called with bigger size * nmemb than length of destination buffer [-Werror]
  return __fread_chk (__ptr, __bos0 (__ptr), __size, __n, __stream);
                                                                  ^
compilation terminated due to -Wfatal-errors.
lto1: all warnings being treated as errors
lto-wrapper: /usr/bin/x86_64-linux-gnu-g++ returned 1 exit status
/usr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[4]: *** [release.linux] Error 1
```

It seems that a long-open gcc bug causes spurious warnings when LTO is enabled: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53431 And the -Werror flag turns these eventually into errors, and thus the build process fails. Unfortunately, there is no way to selectively disable the warning with pragmas or -Wno-error=xxx flags.

ruben-ayrapetyan, 2015-11-12:
Could we instead, add `-Werror=...` for all warnings in list of enabled warnings, like the following: `-Wall -Werror=all -Wextra -Werror=extra ...`, so leaving `Werror` enabled for all warnings that we consider to be significant, and disabling it for the others. Would this disable `Werror` for `pragma` warnings?

akiss77, 2015-11-12:
Yes, being more specific on `-Werror=xxx` can work. Side note: it will be a hell to maintain. I'll update pr #719 accordingly, let's continue discussion on potential solutions there. (If we agree here that the issue should be solved.)

ruben-ayrapetyan, 2015-11-13:
@akiss77, maybe it would be easier to maintain if all `-Werror=` options would be next to corresponding `-W` options, like the following:
```
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wall -Werror=all")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wextra -Werror=extra")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -pedantic -Werror=pedantic")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wformat-nonliteral -Werror=format-nonliteral")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Winit-self -Werror=init-self")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wno-stack-protector -Werror=no-stack-protector")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wconversion -Werror=conversion")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wsign-conversion -Werror=sign-conversion")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wformat-security -Werror=format-security")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wmissing-declarations -Werror=missing-declarations")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wno-attributes -Werror=no-attributes")
set(COMPILE_FLAGS_JERRY "${COMPILE_FLAGS_JERRY} -Wlogical-op -Werror=logical-op")
```

And to not duplicate every warning in `-W` and `-Werror` maybe its better to introduce a cmake routine for registering warnings by their names.

akiss77, 2015-11-15:
@ruben-ayrapetyan If you take a look at the second version of the patch at #719, you'll see that we cannot specify `-Wxxx -Werror=xxx` together since even in the original/current version `-Werror` is given separately and guarded by a check for GNU C/C++ compilers. (A bit of an extra is that some flags don't work with `-Werror=xxx`, e.g., `-Wno-stack-protector` and `-Wno-attributes` are not to be paired with `-Werror=xxx`.) A routine might help indeed.

ruben-ayrapetyan, 2015-11-15:
@akiss77, seems that Werror can't work with `-Wno-stack-protector` and `-Wno-attributes` just because the options are for disabling corresponding warnings, not for enable them. So, for example, `Werror=attributes` should work, isn't it?

A new routine can take warning name and argument that would specify whether to enable or disable the warning. Also, seems that the check for GNU C/C++ compiler can be included into the routine.

Furthermore, we can make routine that accepts not a single warning name, but a list of warning names to be enabled / disabled. Seems that in the case, warning declaration block would be even more compact than it is in current master branch.

akiss77, 2015-11-17:
@ruben-ayrapetyan please see latest version at #719

ruben-ayrapetyan, 2015-11-17:
@akiss77, the latest version looks good.

PR #719 (2015-11-18): Pass -Wno-error to linker in case of LTO builds
When linking a release-built command line shell on Linux against
default glibc with LTO enabled, a long-open gcc bug causes spurious
warning to be emitted around a call to fread. The -Werror flag
turns this into an error and the build process fails.
Unfortunately, there is no way to selectively disable the warning
with pragmas or -Wno-error=xxx flags.

Thus, this patch passes -Wno-error to the linker in case of LTO
builds. (But only to the linker. The compilation still applies the
warnings-are-errors policy.)

fixes #718
closed 2015-11-18 by LaszloLango