jerryscript-project/jerryscript #722: Q. current internal encoding for jerry ?

chunseoklee, 2015-11-19:
I am working on encoding stuff for IoT.js.
I have a question on jerry encoding.  
For example, ```lit_code_point_to_cesu8```  function seems to return UTF8 string although function name and comments has cesu8 encoding. 
Here is my question : which encoding is used  for ```lit_utf8_byte_t *``` ?  cesu8 or utf8 ?

dbatyai, 2015-11-19:
Strings are encoded internally as cesu8, even though it says utf8 in function names. See discussion about this [here](https://github.com/Samsung/jerryscript/pull/616#issuecomment-142548923).

chunseoklee, 2015-11-19:
@dbatyai Thanys for help. 
I have 1 more q. 
``` ecma_string_to_utf8_string ````  returns utf8 string or cesu8 string ?

dbatyai, 2015-11-20:
@chunseoklee, `ecma_string_to_utf8_string` returns a cesu8 encoded string as well.

chunseoklee, 2015-11-23:
@dbatyai  It is strange. ```jerry_api_string_to_char_buffer``` for "\uD834\uDD1E(MUSICAL SYMBOL G CLEF' (U+1D11E))" seems to return data of 4 byte lengths which is illegal for cesu-8 encoding.

chunseoklee, 2015-11-23:
@dbatyai I found the reason why. Current iot.js's jerry submodule is on commit 08c6. But cesu8 patch were merged later.

chunseoklee, 2015-12-02:
On latest commit, ```jerry_api_string_to_char_buffer``` returns cesu encoded format string. 
I will close this issue.

closed 2015-12-02 by chunseoklee