jerryscript-project/jerryscript #735: how to build jerry standalone for stm32f4/nuttx

jiangzidong, 2015-11-24:
There is branch for stm32f4 in makefile, but do we have wiki to describe the steps to build/install jerry on stm32f4/nuttx like https://github.com/Samsung/iotjs/wiki/Build-for-Nuttx-(new)

egavrin, 2015-11-24:
`make release.mcu_stmf4-cp -j` this will build version for stmf4.

jiangzidong, 2015-11-24:
@egavrin for iotjs, Nuttx need some special configuration/customization. what about jerry, does it has any requirement for the Nuttx config?

egavrin, 2015-11-24:
We have no NuttX integration, we only build bare metal firmware for STM.

jiangzidong, 2015-11-24:
@egavrin Sorry, I don't understand. How can we use the "bare firmware" in STM? Could you give me an example?

egavrin, 2015-11-24:
This means that firmware contains no OS, only Jerry itself ( see
./jeery-core/main-mcu.cpp) and JavaScript file to be executed is built-in
into binary and cannot be changed .

seanshpark, 2015-11-24:
Hi, you guys are already talking about this. I would like to prepare JerryScript in NuttX/STM32F4. I think adding to `embedding` seems nice(though might take a while to land) @egavrin , is it ok to add pages to WiKi or should I update web pages? I'd prefer WiKi :)

egavrin, 2015-11-24:
@seanshpark wiki is OK :)

seanshpark, 2015-11-25:
I think `readme.md` file can be one way. I've made [embedding-nuttx](https://github.com/Samsung/jerryscript/tree/embedding-nuttx/embedding/nuttx) branch so to work on for the moment. After #733 lands I'll make a new PR to merge NuttX... 
or @egavrin , maybe merge `embedding-nuttx` to #733 now?
@jiangzidong , please use this branch for the moment. Add questions if need assist.

LaszloLango, 2016-02-11:
Related PR: #805

closed 2016-07-28 by zherczeg