jerryscript-project/jerryscript #737: ICE: Assertion 'lit_cp.packed_value != MEM_CP_NULL' opcodes.cpp(opfunc_func_expr_n):602

egavrin, 2015-11-24:
Version: 17a5dd27bbbadc036d83be0448f9edfa90800147

System:
```
Linux eg 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Output:
```

ICE: Assertion 'lit_cp.packed_value != MEM_CP_NULL' failed at /home/eg/dev/jr/jerry-core/vm/opcodes.cpp(opfunc_func_expr_n):602.
Error: ERR_FAILED_INTERNAL_ASSERTION

```
Testcase:

```js
 new function f(f) {
  return {className: 'xxx'};
};
x = 1;
function g(active) {
  for (i = 1; i <= 1000; i++) { if (i == active) {
  x = i;   if (f("" + i) != null) { }
    } else {
  if (f("" + i) != null) }
  }
}
g(0)
```

```js
 new function a(a) {;for (f in [1,2,3]) print(f);
}; 1;
function g(active) {
  for (i = 1; i <= 1000; i++) { if (i == active) {
  xI
      if (f != null) { }
    } else {
  if (f("" + i) != null) }
  }
}
g(0)
```

```js
 new function f(f) {;for (f in [1,2,3]) pRint(f);
}; 1;
function g(active) {
  for (i = 1; i <= 1000; i++) { if (i == active) {
  x 
      if (f != null) { }
    } else {
  if (f("" + i) != null) }
  }
}
g(0)
```

LaszloLango, 2016-02-08:
This fails with `PARSE_ERROR: Statement expected. (line: 9, column: 26)` on the current master (91a0514). @egavrin, @zherczeg is this the expected result?

zherczeg, 2016-02-09:
Of course.

This is not a valid statement:

{ if (f("" + i) != null) /* statement is missing here */ }

closed 2016-02-09 by zherczeg