jerryscript-project/jerryscript #742: ICE: Assertion 'buff_index <= buff_size' failed at (ecma_builtin_number_prototype_object_to_string):309

egavrin, 2015-11-24:
Version: 17a5dd27bbbadc036d83be0448f9edfa90800147

System:
```
Linux eg 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Output:
```

ICE: Assertion 'buff_index <= buff_size' failed at /home/eg/dev/jr/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp(ecma_builtin_number_prototype_object_to_string):309.
Error: ERR_FAILED_INTERNAL_ASSERTION

```
Testcase:

```js
// Cj
var big = 2147483646;

big++;
assert(big == 2147483647);

big += 1;
assert(big == 2147483648); // s

big++;
assert(big == 2147483649); // s

assert ((1152921504606846600).toString([1,2,3,4].slice(1,2)) === "11506847000")

rt (1.79e+308 === Infinity);
```

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

Minimal testcase:
`assert ((1152921504606846600).toString([1,2,3,4].slice(1,2)) === "11506847000");`

ktorpi, 2016-07-11:
Still fails on current master (9bce5b0).

akosthekiss, 2016-08-29:
The issue is still valid on current master (8edab96).

tilmannOSG, 2017-02-24:
Still fails on master (62a581f).  
  
`$ build/bin/jerry 
jerry> assert ((1152921504606846600).toString([1,2,3,4].slice(1,2)) === "11506847000");
ICE: Assertion 'buff_index <= buff_size' failed at /home/t/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.c(ecma_builtin_number_prototype_object_to_string):409.
Error: ERR_FAILED_INTERNAL_ASSERTION`

zherczeg, 2017-02-27:
I have no idea how to fix this. Problem is:

log (115292150460684000) / log (2) = 59.999999999999999529496732252183888148671846990644

floor(log (115292150460684000) / log (2)) -> should be 59.

However, it is 60 with double precision.

robertsipka, 2017-03-09:
@zherczeg : That's right! The problem comes from the inaccuracy of the log (115292150460684660) / log (2) division. I've added a PR above, which fixes it.

robertsipka, 2017-03-20:
Fixed in #1650.

closed 2017-03-20 by robertsipka