jerryscript-project/jerryscript #744: ICE: Assertion 'end <= ecma_string_get_length (appended_string_p)' failed at (ecma_builtin_string_prototype_object_replace_append_substr):776

egavrin, 2015-11-24:
Version: 17a5dd27bbbadc036d83be0448f9edfa90800147

System:
```
Linux eg 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Output:
```

ICE: Assertion 'end <= ecma_string_get_length (appended_string_p)' failed at /home/eg/dev/jr/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.cpp(ecma_builtin_string_prototype_object_replace_append_substr):776.
Error: ERR_FAILED_INTERNAL_ASSERTION

```
Testcase:

```js
 '஻BBCB@abXde^'.replace(/|/g,"$'஻BBBB@abXde^ce((/aq");
```

```js
 '஻BB8B@abXde^'.replace(/a/g,"$஻Bce((/a%");
```

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

LaszloLango, 2016-04-19:
Minimized testcase: `''.replace(/|/g,"஻");`

akosthekiss, 2016-08-29:
The issue is still valid on current master (8edab96).

LaszloLango, 2016-09-08:
#1332 is landed

closed 2016-09-08 by LaszloLango