jerryscript-project/jerryscript #754: How to get a function object using API

qdk0901, 2015-11-29:
Hi,
  I read the API reference but can not figure out how to get a function object using the API, for example, there is piece of js code
'''
function testFunction() {
   //.....
}
'''
I want to know how to get the 'testFunction' from C code using API directly.
Many thanks.

qdk0901, 2015-11-29:
figure out now, below code works fine
```
	jerry_run ();
	
	jerry_api_value_t val2;
	jerry_api_value_t val3;
	bool is_ok = jerry_api_get_object_field_value (glob_obj_p, (jerry_api_char_t*)"testFunction2", &val2);
	
	if (is_ok) {
		jerry_api_call_function(val2.v_object, NULL, &val3, NULL, 0);
		jerry_api_release_object (val2.v_object);
	}
```

ruben-ayrapetyan, 2015-11-30:
@qdk0901, although, in this particular example, type of value in `val2` would always be object, the `jerry_api_get_object_field_value` could also receive values of another types. So, its better to explicitly check for type of the received values through `type` field, or to add an `assert` that type is same as expected.

closed 2015-12-23 by ruben-ayrapetyan