jerryscript-project/jerryscript #755: Please vote for logo

Seo-Young, 2015-11-30:
We've got diverse logo candidates. 
Now, it's time to choose. :)
Please vote for JerryScript as follow. 
  - Give comment with the number of candidate.
  - You can pick two logo candidates up. 
  - This voting will be closed at 7:00 UTC on 12/4(Friday)

1. [image]
[image]
[image]
   - This has some iterations.
   - This has one issue on Jerry. I haven't get the confirmation on use from Warner Bros.(still on waiting.... ;( )

2. [image]

3. [image]

4. [image]

5. [image]

6. [image]

7. [image]

8. [image]
[image]
   - This has some iterations.

chunseoklee, 2015-11-30:
4

Seo-Young, 2015-11-30:
8

egavrin, 2015-11-30:
5

fivdi, 2015-11-30:
Although I think 8 looks the best, I'd pick the 3rd variant of 1 due to its clear association with JavaScript.

seanshpark, 2015-11-30:
1, 8

lemmaa, 2015-11-30:
8-4

in 

0
1,2,3,4,5,6

sand1k, 2015-11-30:
1, 8

zherczeg, 2015-11-30:
8

LaszloLango, 2015-11-30:
8

ruben-ayrapetyan, 2015-11-30:
8 (version 5)

renatahodovan, 2015-11-30:
8

tczene, 2015-11-30:
8

bzsolt, 2015-11-30:
8

akiss77, 2015-11-30:
8 [bottom line 3rd from left], 4
(extra vote on the third place: 1 [topmost])

minjung978, 2015-12-01:
8

wateret, 2015-12-01:
8

Boxie5, 2015-12-01:
6

rapportyou, 2015-12-01:
5 looks good

rtakacs, 2015-12-01:
8

lvidacs, 2015-12-01:
8 (5th version in the bottom line) and 6

Seo-Young, 2015-12-02:
Share current status.
[image]

idkiller, 2015-12-03:
8 :+1:

Seo-Young, 2015-12-04:
Time's up! 

Now, you can see the result. :)  
[image]

But the selective one has variations so we need following decision that which one will be the main logo. 
[image]
With these, non selected can be used as iterations for the project. But if you would like to suggest remove any image from iterations, feel free to talk.

Seo-Young, 2015-12-04:
1

idkiller, 2015-12-04:
6 :+1:

fivdi, 2015-12-04:
1

akiss77, 2015-12-04:
3

zherczeg, 2015-12-04:
6 we need the tail :)

ruben-ayrapetyan, 2015-12-04:
4, 5

egavrin, 2015-12-04:
4,5

Noooo, tail looks ugly!

fivdi, 2015-12-04:
@ruben-ayrapetyan @egavrin during the second round of voting it's only possible to vote for one logo :)

davidsz, 2015-12-04:
Maybe it's a bit late to clarify, but we can easily combine these iterations. For example, I would prefer the first logo without tail:
[image]

rtakacs, 2015-12-04:
8

LaszloLango, 2015-12-04:
4,5

bzsolt, 2015-12-04:
3

lemmaa, 2015-12-05:
7

wateret, 2015-12-05:
3,4

seanshpark, 2015-12-06:
7

Seo-Young, 2015-12-07:
@ruben-ayrapetyan @egavrin @LaszloLango @wateret Please re-vote. In this turn, you can pick only one. 
@davidsz So, is your decision 7? without tail?
@lemmaa You picked up 7, correct?

When everything are cleared. I will close this issue.

lemmaa, 2015-12-07:
@Seo-Young Yes, 7!

ruben-ayrapetyan, 2015-12-07:
5

wateret, 2015-12-07:
3

davidsz, 2015-12-07:
@Seo-Young Yes, 7 :)

egavrin, 2015-12-07:
7

LaszloLango, 2015-12-07:
4

sand1k, 2015-12-07:
4

Seo-Young, 2015-12-07:
Here's the result. :)
1 and 8 are same one so integrated.
[image]
If anyone would like to change your mind, please comment until today UTC 23:00. If not, we need other following voting.

seanshpark, 2015-12-08:
7 --> 1
I like the tail :)

wateret, 2015-12-08:
~~3~~ -> 4

Sorry for the tail.

ruben-ayrapetyan, 2015-12-08:
5 -> 4

zherczeg, 2015-12-08:
my only problem with 4 that people will think this is the moon js :)

seanshpark, 2015-12-08:
> moon js :)

um... how about cut the cheese in half and fill only half of the circle, with some tilt, like the one filled S in 1 and 7

zherczeg, 2015-12-08:
or making it square with holes perhaps

seanshpark, 2015-12-08:
https://www.youtube.com/watch?v=T0qagA4_eVQ 
moon is cheese :)

lemmaa, 2015-12-08:
FYI

In history of JerryScript, cheese is used for naming the IR, originate from @egavrin 's document, the `high-fat cheese IR` and the `low-fat cheese IR`. So, cheese inside 'JS' is reasonable for me, but 'JS' on the cheese is not. :)

egavrin, 2015-12-08:
7

idkiller, 2015-12-08:
In my Opinion, thick border makes the cheese to strange.
I think If whole JS  glyph is cheese, it looks better.

Seo-Young, 2015-12-08:
[image]
The winner is 4! :dango: 
With above comments, I draw some variations. Which one is better?
[image]

seanshpark, 2015-12-08:
Yeah~~

> Which one is better?

the first one. (second has a bite which reminds of some company)

Seo-Young, 2015-12-09:
Additionally, I suggest to release our logo under MIT License, when it's finally decided. 
Please let me know your opinions.

[51 more replies]

closed 2015-12-24 by Seo-Young