jerryscript-project/jerryscript #768: Performance comparison between JerryScript vs Duktape

niklauslee, 2015-12-07:
I'm looking for JavaScript impl for my wearable projects using ARM Cortex-M4 MCUs. JerryScript and Duktape are final candidates and I especially interested in JerryScript because it consumes very low memory (My MCUs has 128KB RAM). I found a benchmark about this: http://samsung.github.io/jerryscript/benchmark/benchmark.html

Could you please let me know what optimization settings were applied to Duktape? Did you applied default memory setting, otherwise what kinds of DUK_OPT_* were applied?

wateret, 2015-12-08:
The results are from default build of `duktape`. Are there any options that must be applied? There might be some options that are essential or really important for benchmark.

wateret, 2015-12-08:
While searching for duktape, I found that [official duktape performance page](http://wiki.duktape.org/Performance130.html) uses `gcc -O2`.

So I experimentally measured sunspider on RPi2.
What's interesting is `bitops-bitwise-and.js` got slower by 14%.

`duk` -> `duk.O2`

                               Benchmark |         RSS<br>(+ is better) |        Perf<br>(+ is better) |
                               --------- |                          --- |                         ---- |
                              3d-cube.js |          196->   196 (0.000) |         1.04-> 0.992 (4.615) | 
                  access-binary-trees.js |          212->   212 (0.000) |        1.184->  1.12 (5.405) | 
                      access-fannkuch.js |          152->   152 (0.000) |        1.908-> 1.856 (2.725) | 
                         access-nbody.js |          160->   160 (0.000) |        1.672-> 1.596 (4.545) | 
             bitops-3bit-bits-in-byte.js |          148->   148 (0.000) |         0.42-> 0.404 (3.809) | 
                  bitops-bits-in-byte.js |          148->   148 (0.000) |        0.696->  0.66 (5.172) | 
                   bitops-bitwise-and.js |          148->   148 (0.000) |      7.204-> 8.236 (-14.325) | 
                controlflow-recursive.js |          192->   192 (0.000) |        1.18-> 1.048 (11.186) | 
                          math-cordic.js |          152->   152 (0.000) |        2.428-> 2.288 (5.766) | 
                    math-partial-sums.js |          148->   148 (0.000) |        2.952-> 2.924 (0.949) | 
                   math-spectral-norm.js |          152->   152 (0.000) |        1.044->  1.02 (2.299) | 
                         string-fasta.js |          164->   164 (0.000) |        4.536-> 4.304 (5.115) | 
                         Geometric mean: |            RSS reduction: 0% |            Speed up: 3.2651% |

% Binary size: 170K -> 198K.

niklauslee, 2015-12-08:
Thank you for your answer.

I meant "build options", not compiler optimization options. There are many build options for Duktape to reduce memory usage as described in https://github.com/svaarala/duktape/blob/master/doc/low-memory.rst. For example, turning off support of regex, referencing counting, or using pointer compression. According to Duktape homepage(http://duktape.org/, See the Code and RAM footprint section), startup RAM (70KB) could be reduced to 24KB with full lowmem options in x86 target.

lemmaa, 2015-12-08:
@niklauslee I think comparing JerryScript to Duktape with 'build option' is not good idea. Because they are different in feature. For JerryScript, we also support a build option 'compact profile' which results much smaller footprint with limited features of ECMA-262 standard. It can run on only 8KB start up RAM. In conclusion, such an approach depends on what we should support or what we shouldn't.

By the way, which features are ignorable to you in JavaScript for wearable device? For example, regex, UTF-16, etc. Your opinion will be a good reference to us.

niklauslee, 2015-12-08:
I've just did heap profiling of Duktape with some optimization options. I was able to reduce RAM usage from 135KB to 103KB with "hello" example. I think Duktape is not good for my 128KB MCU. JerryScript looks much better than Duktape in the aspect of memory consumption. I will try to port JerryScript to my MCU from now.

@lemmaa I didn't know JerryScript also provides such build options. Sounds good! I will not use regex and unicode code support. It is preferred to be able to disable each of features using build options. Thanks!

seanshpark, 2015-12-08:
> I will not use regex and unicode code support. It is preferred to be able to disable each of features using build options

@egavrin , what do you think about this? I give a +1 to support this for small devices.

egavrin, 2015-12-09:
@seanshpark more configurability is good, so :+1: too. As you know, currently, it's possible for Compact Profile, by editing `./jerry-core/config.h`. The only thing left - is to enable build options for this and enable for full profile. See #88 -  we have small technical debt on this.

poussa, 2016-01-29:
Can we add duktape low memory data to the graphs? That is, build duktape with low memory options and add that data to the .json files. We should do the same for JerryScript: default build and compact profile. This way we can compare the default and low memory configurations of both engines. 

How are the .json files generated? I could not find any tools for that in the src tree.

svaarala, 2016-01-29:
@niklauslee Just to summarize Duktape minimum RAM usage:

- With Duktape [low memory options](https://github.com/svaarala/duktape/blob/master/doc/low-memory.rst) startup RAM usage is around 27kB at the moment
- There's a pull request (https://github.com/svaarala/duktape/pull/559 to be merged this week) which brings startup RAM usage to around 3-4kB by moving almost all builtin objects and strings into ROM (read only data section)

To achieve these config options need to be tuned for low memory, and the figures above assume "pointer compression" is used (in essence packing pointers into 16-bit integers).

niklauslee, 2016-01-30:
@svaarala Thank you for informing. The improvement looks very attractive for low memory MCUs. I'll try Duktape once again when the pull request is merged.

zherczeg, 2016-01-31:
I think initial memory consumption is just one aspect of an engine. Runtime memory consumption is also important, and we constantly improves that as well. For example, the JerryScript team has just finished the development of Compact Byte Code, which is a variable length byte code with lightweight data compression called opcode morphing. It combines high performance (15% faster) with small byte-code size (around 50% saving). We are very proud of this achievement and I think it is worth to publish a paper about our findings.

svaarala, 2016-01-31:
@zherczeg I agree, the two important basic factors are (1) initial RAM usage and (2) how RAM usage grows with program size. The less memory you have the more important (1) becomes because it determines whether there will be space for user programs at all; the more memory you have the more important (2) then becomes. So far most work in Duktape has been for (1) because it's been the bottleneck issue for very low memory targets (64kB or less), and there's much work ahead for (2). Variable length bytecode sounds interesting, bytecode is indeed a major memory usage factor.

closed 2016-07-28 by zherczeg