jerryscript-project/jerryscript #770: Idea for API improvement

egavrin, 2015-12-08:
> I did a little exploration in jerryscript and IoT.js and noticed that IoT.js uses jerry_api_value_t structure defined in JerryScript API by accessing its members directly to do type checking and value retrieval activities. It is done through fuction-like macros defined in iotjs_binding.cpp: JVAL_IS_NULL, JVAL_IS_STRING, JVAL_TO_NUMBER, etc.
What is your opinion about this solution?
Maybe, it would be worthwhile to refine the JerryScript API by providing type checker and value retrieval functions to struct jerry_api_value_t in order to avoid the direct access of its internal structure in IoT.js. This can result clearer and more maintainable code that facilitates the future developments.

by @ktorpi

lemmaa, 2015-12-08:
+1, https://github.com/Samsung/jerryscript/issues/711

LaszloLango, 2016-02-09:
Related PRs: #788, #810

closed 2016-07-14 by LaszloLango