jerryscript-project/jerryscript #772: "scopes_tree_count_literals_in_blocks", a bug or on purpose?

jiangzidong, 2015-12-09:
The scopes_tree_count_literals_in_blocks in `scopes-tree.cpp` is called by  `serializer_merge_scopes_into_bytecode`. It is used to count slots needed for a scope's hash table, and **should**  divide bytecode into blocks and count unique literal indexes used in each block, as the code comment says. 

However, in its implementation, the global_oc will never increase and next_uid will become 0 when enter a subscope, so **it becomes counting the unique literal used in one instruction, instead of one block.**. So its result will be larger, and leads to memory waste in lit-id-hash-table.

**I fixed it** (https://github.com/jiangzidong/jerryscript/tree/litcount), but the run-perf-result is not so good

                               Benchmark |         RSS<br>(+ is better) |        Perf<br>(+ is better) |
                               --------- |                          --- |                         ---- |
                              3d-cube.js |          132->   132 (0.000) |       0.344->0.3496 (-1.628) |
                  access-binary-trees.js |          84->    72 (14.286) |       0.164->0.1688 (-2.927) |
                      access-fannkuch.js |          48->    52 (-8.333) |       0.912->0.9192 (-0.789) |
             bitops-3bit-bits-in-byte.js |          40->    36 (10.000) |       0.276->0.2784 (-0.870) |
                  bitops-bits-in-byte.js |           40->    40 (0.000) |      0.3616->0.3632 (-0.442) |
                   bitops-bitwise-and.js |           40->    40 (0.000) |       0.344-> 0.348 (-1.163) |
                controlflow-recursive.js |          244->   240 (1.639) |       0.152-> 0.156 (-2.632) |
                           crypto-aes.js |          152->   152 (0.000) |       0.604->0.6088 (-0.795) |
                           crypto-md5.js |         204->   208 (-1.961) |        3.74->3.8976 (-4.214) |
                          crypto-sha1.js |          148->   148 (0.000) |      1.6408-> 1.684 (-2.633) |
                    date-format-tofte.js |          112->   112 (0.000) |        0.384->  0.36 (6.250) |
                    date-format-xparb.js |           96->    92 (4.167) |        0.18-> 0.184 (-2.222) |
                          math-cordic.js |           48->    48 (0.000) |       0.3856->0.3856 (0.000) |
                   math-spectral-norm.js |           44->    44 (0.000) |          0.2->   0.2 (0.000) |
                        string-base64.js |          200->   200 (0.000) |     28.4536->30.448 (-7.009) |
                         string-fasta.js |           60->    60 (0.000) |       0.624->0.6432 (-3.077) |
                         Geometric mean: |       RSS reduction: 1.3603% |            Speed up: -1.474% |
 
I also measure the heap peak counts:

 Benchmark | heap peak counts old | heap peak counts new | + is better |
  ----------|---------------|----------------|--------------|
 date-format-tofte.js |   636 |    636 |     0.000000 |
 crypto-aes.js|   1164|    1174|    -0.859107|
 string-base64.js|        2577|    2575|    0.077610|
 string-fasta.js| 223|     223|     0.000000| 
 math-spectral-norm.js|   130|     130|     0.000000|
 math-cordic.js|  130|     129|     0.769231|
 date-format-xparb.js|    562|     556|     1.067616|
 access-binary-trees.js|  727|     727|     0.000000|
 bitops-bits-in-byte.js|  67|      67|     0.000000|
 crypto-md5.js|   2293|    2291|    0.087222|
 3d-cube.js|      646|     644|     0.309598|
 bitops-3bit-bits-in-byte.js|     73|      73|      0.000000|
 bitops-bitwise-and.js|   50|      49|      2.000000|
 controlflow-recursive.js|        723|     722|     0.138313|
 access-fannkuch.js|      99|      99|      0.000000|
 crypto-sha1.js|  1461|    1459|    0.136893|
 Geometric mean| | | 0.23%|

I think the reason maybe: 
 * memory: the hash_table for each blocks is larger than before. it need to store all literals used in code block, rather than one instruction. This will counteract the decrease of the lid-id-hash-table
 * performance:  during serializer_merge_scopes_into_bytecode, more items in hash_table leads to more look_up actions.

But I still believe it is a bug in scopes_tree_count_literals_in_blocks. Should I create a PR about that?

ruben-ayrapetyan, 2015-12-09:
@jiangzidong, thank you for your question and detailed investigation. Yes, it is a bug.
The literal tables mechanism is planned to be reworked in near future, so the bug should also become fixed soon.

ruben-ayrapetyan, 2015-12-23:
@jiangzidong, mechanism of literal tables construction was significantly reworked in PR #789. The bug is fixed in the new version - there is no `global_oc` global instruction counter now.

The PR was already merged, so closing the issue.

closed 2015-12-23 by ruben-ayrapetyan