jerryscript-project/jerryscript #774: Unittests build fails when LTO is disabled after #756

bzsolt, 2015-12-09:
```LTO=OFF make unittests``` produces the following errors:
```
jerryscript/jerry-core/jerry.cpp:1409: undefined reference to `jerry_port_errormsg'
jerryscript/jerry-core/jerry.cpp:1398: undefined reference to `jerry_port_errormsg'
jerry-core/libunittests.jerry-core.a(lit-strings.cpp.o): In function `lit_put_ecma_char(unsigned short)':
jerryscript/jerry-core/lit/lit-strings.cpp:1013: undefined reference to `jerry_port_putchar'
jerryscript/jerry-core/lit/lit-strings.cpp:1008: undefined reference to `jerry_port_putchar'
jerry-core/libunittests.jerry-core.a(parser.cpp.o): In function `parse_property_name':
jerryscript/jerry-core/parser/js/parser.cpp:351: undefined reference to `jerry_port_putchar'
jerry-core/libunittests.jerry-core.a(parser.cpp.o): In function `current_token_must_be':
jerryscript/jerry-core/parser/js/parser.cpp:132: undefined reference to `jerry_port_putchar'
jerry-core/libunittests.jerry-core.a(parser.cpp.o): In function `jsp_find_next_token_before_the_locus':
jerryscript/jerry-core/parser/js/parser.cpp:272: undefined reference to `jerry_port_putchar'
```

egavrin, 2015-12-09:
#756

seanshpark, 2015-12-10:
Sorry for build break...

bzsolt, 2015-12-16:
Fixed in [fe2a76f](https://github.com/Samsung/jerryscript/commit/fe2a76fcedb62cbdb042ca30d391fe771c3703a4)

closed 2015-12-16 by bzsolt