jerryscript-project/jerryscript #776: Why is BLOCK_SIZE 64? Can it be 128?

jiangzidong, 2015-12-10:
Now the BLOCK_SIZE in `bytecode-data.h` is defined `64u`. While the idx range for literal is 0-127. So I think the logic behind the decision is that **one opcode has at most 2 literal values**.

However, in practice, I found one opcodes has atmost **ONE** literal values. Am I right?
I change the BLOCK_SIZE to 128, and all test case passed.

ruben-ayrapetyan, 2015-12-10:
@jiangzidong, actually, an instruction can contain more than one literal value. There is no "1 literal per instruction" restriction. Actually, an instruction can contain up to 3 literal values. So, seems that there is a bug - even 64 value is too big.

jiangzidong, 2015-12-10:
@ruben-ayrapetyan could you give me an example? in my practice, there is always 0-1 lit in one opcode

ruben-ayrapetyan, 2015-12-10:
Maybe, there are no examples in currently generated byte-code, but currently nothing prevents a pull request with change in byte-code generator to produce more optimal byte-code. For example:
```
addition c, a, b
```

instead of

```
assignment tmp131, 6, a
addition tmp132, tmp131, b
assignment c, 6, tmp132
```

jiangzidong, 2015-12-10:
@ruben-ayrapetyan I see, thanks for your explaination!

closed 2015-12-10 by jiangzidong