jerryscript-project/jerryscript #780: libc builtin vs system mixup

obastemur, 2015-12-10:
Working on OSX, builtin libc is pretty mixin up with the one from system. I could contribute this part though but have some questions first;

 - preferred libc target should be the one built-in right ?  IMHO Yes.
 - for the sake of testing / stability etc. why not separate jerry_libc project ?
 - main_linux, main_darwin etc. could benefit lots of improvements and they could be combined into a single file. Agree / disagree ?

egavrin, 2015-12-10:
Your contribution to libc would be really helpful to us, thank you!

- Yes, preferred libc should be the built-in one.
- We thought about separating jerry-libc, but for now it's not mature enough to be a separate project, or to be used with other projects like IoT.js. From the view of testing - currently, all default testing environments of the engine have jerry-libc, so any commit tested by developer or by our CI checks the libc.
- Absolutely, `main_*` should have at commons part that can be shared across the platforms.

seanshpark, 2015-12-10:
I'm also interested in this subject, and agree with @egavrin

LaszloLango, 2016-02-25:
* after #331 is merged, you are able to use the compiler's default libc
 * #819 is also merged, so there are no separate `main_linux` and `main_darwin` files.

The tests are currently failing on OSX, so any help with fixing it is welcome. For details see our travis builds: https://travis-ci.org/Samsung/jerryscript

closed 2016-02-25 by LaszloLango