jerryscript-project/jerryscript #781: Assertion '!ecma_is_value_empty (reg_value)' failed in get_variable_value

renatahodovan, 2015-12-14:
###### Jerry version:
    Checked revision: 2c3c952
    Build: debug.linux
###### OS: 
    Ubuntu 15.10, x86_64
###### Test case:
```javascript
try { Function(Array.isArray) } catch (err) { }
try { Function(TypeError.prototype) } catch (err) { }
```

###### Backtrace:
```

(function(){function(){/* ecmascript */}})
                    ^
ERROR: Ln 1, Col 21: Expected 'Identifier' token
ICE: Assertion '!ecma_is_value_empty (reg_value)' failed at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp(get_variable_value):85.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004afcc5 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004afcc5 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=11651, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000437141 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:67
# 4  0x0000000000437190 in jerry_assert_fail (assertion=0x4d4240 "!ecma_is_value_empty (reg_value)", 
    file=0x4d4198 "jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp", 
    function=0x4b1780 <_ZZ18get_variable_valueP14vm_frame_ctx_thbE8__func__.lto_priv.1111> "get_variable_value", line=85)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:99
# 5  0x000000000049ffcc in get_variable_value (frame_ctx_p=0x7fffffffc800, var_idx=128 '\200', do_eval_or_arguments_check=false)
    at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp:85
# 6  0x000000000043544a in opfunc_retval (instr=..., frame_ctx_p=0x7fffffffc800)
    at jerryscript/jerry-core/vm/opcodes.cpp:1174
# 7  0x00000000004a7f45 in vm_loop (frame_ctx_p=0x7fffffffc800, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:545
# 8  0x00000000004a81da in vm_run_from_pos (header_p=0x8e7b00 <mem_heap+1984>, start_pos=1, this_binding_value=523, lex_env_p=0x8e7760 <mem_heap+1056>, 
    is_strict=false, is_eval_code=true, arg_collection_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:643
# 9  0x00000000004a7df9 in vm_run_eval (bytecode_data_p=0x8e7b00 <mem_heap+1984>, is_direct=false)
    at jerryscript/jerry-core/vm/vm.cpp:484
# 10 0x0000000000448a58 in ecma_op_eval_chars_buffer (code_p=0x926680 <mem_heap+258880> "(function(){TypeError})/* ecmascript */}})", code_buffer_size=23, 
    is_direct=false, is_called_from_strict_mode_code=false) at jerryscript/jerry-core/ecma/operations/ecma-eval.cpp:113
# 11 0x00000000004488f7 in ecma_op_eval (code_p=0x8e7ad0 <mem_heap+1936>, is_direct=false, is_called_from_strict_mode_code=false)
    at jerryscript/jerry-core/ecma/operations/ecma-eval.cpp:63
# 12 0x000000000047d420 in ecma_builtin_function_dispatch_construct (arguments_list_p=0x9266c0 <mem_heap+258944>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-function.cpp:203
# 13 0x000000000047ce2d in ecma_builtin_function_dispatch_call (arguments_list_p=0x9266c0 <mem_heap+258944>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-function.cpp:56
# 14 0x000000000049bed9 in ecma_builtin_dispatch_call (obj_p=0x8e78d8 <mem_heap+1432>, this_arg_value=4, arg_collection_p=0x8e79e8 <mem_heap+1704>)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:116
# 15 0x000000000044a107 in ecma_op_function_call (func_obj_p=0x8e78d8 <mem_heap+1432>, this_arg_value=4, arg_collection_p=0x8e79e8 <mem_heap+1704>)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:922
# 16 0x00000000004346d1 in opfunc_call_n (instr=..., frame_ctx_p=0x7fffffffcd60) at jerryscript/jerry-core/vm/opcodes.cpp:808
# 17 0x00000000004a7f45 in vm_loop (frame_ctx_p=0x7fffffffcd60, run_scope_p=0x7fffffffccb0)
    at jerryscript/jerry-core/vm/vm.cpp:545
# 18 0x00000000004a030f in opfunc_try_block (instr=..., frame_ctx_p=0x7fffffffcd60)
    at jerryscript/jerry-core/vm/opcodes-ecma-try-catch-finally.cpp:41
# 19 0x00000000004a7f45 in vm_loop (frame_ctx_p=0x7fffffffcd60, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:545
# 20 0x00000000004a81da in vm_run_from_pos (header_p=0x8e7800 <mem_heap+1216>, start_pos=1, this_binding_value=523, lex_env_p=0x8e7760 <mem_heap+1056>, 
    is_strict=false, is_eval_code=false, arg_collection_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:643
# 21 0x00000000004a7c3e in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:415
# 22 0x00000000004aaed9 in jerry_run () at jerryscript/jerry-core/jerry.cpp:1533
# 23 0x00000000004a96b1 in main (argc=3, argv=0x7fffffffd388) at jerryscript/main-linux.cpp:459
```

LaszloLango, 2016-02-08:
Fails on current master (91a0514) with the following parser error, but runs fine in chrome console.
`PARSE_ERROR: Identifier expected. (line: 1, column: 21)`
@renatahodovan, What is the expected result of this test?

renatahodovan, 2016-02-08:
This is a fuzzer-generated test. Its only expected result is not to fail with an assert.

LaszloLango, 2016-02-08:
Ok, then it works on master.

closed 2016-02-08 by LaszloLango