jerryscript-project/jerryscript #787: Assertion 'mem_free_chunks_number == 0' failed in mem_pools_finalize

renatahodovan, 2015-12-14:
###### Jerry version:
    Checked revision: 2c3c952
    Build: debug.linux
###### OS: 
    Ubuntu 15.10, x86_64
###### Test case:
```javascript
Array.prototype.push(Math.sin);
Object.freeze(Array.prototype);
String.prototype.match(String.prototype);
```

###### Backtrace:
```
ICE: Assertion 'mem_free_chunks_number == 0' failed at jerryscript/jerry-core/mem/mem-poolman.cpp(mem_pools_finalize):192.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004afcc5 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004afcc5 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=10775, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000437141 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:67
# 4  0x0000000000437190 in jerry_assert_fail (assertion=0x4d601d "mem_free_chunks_number == 0", 
    file=0x4d5fd0 "jerryscript/jerry-core/mem/mem-poolman.cpp", 
    function=0x4b12f0 <_ZZ18mem_pools_finalizevE8__func__.lto_priv.1142> "mem_pools_finalize", line=192)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:99
# 5  0x00000000004a72b2 in mem_pools_finalize () at jerryscript/jerry-core/mem/mem-poolman.cpp:192
# 6  0x00000000004a711a in mem_finalize (is_show_mem_stats=false) at jerryscript/jerry-core/mem/mem-allocator.cpp:52
# 7  0x00000000004aada9 in jerry_cleanup () at jerryscript/jerry-core/jerry.cpp:1439
# 8  0x00000000004a96b9 in main (argc=3, argv=0x7fffffffd368) at jerryscript/main-linux.cpp:465
```

egavrin, 2015-12-17:
Mem leak

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

closed 2016-02-09 by LaszloLango