jerryscript-project/jerryscript #796: How to run test in RPI2

jiangzidong, 2016-01-05:
Hi, what your recommended way to build jerry and run test for RPI2?
1, directly git clone codes on RPI2, then make, then run-perf-test
or
2, cross-build the jerry
or
other ways?

seanshpark, 2016-01-05:
I'm also interested on this and hope to add a document in 'docs' folder.

jiangzidong, 2016-01-06:
and it fails when i directly `make release.linux` on RPI2

```
/usr/include/arm-linux-gnueabihf/bits/byteswap.h: In function ‘unsigned int __bswap_32(unsigned int)’:
/usr/include/arm-linux-gnueabihf/bits/byteswap.h:44:31: sorry, unimplemented: Thumb-1 hard-float VFP ABI
 __bswap_32 (unsigned int __bsx)
                               ^
compilation terminated due to -Wfatal-errors.
```

seanshpark, 2016-01-06:
For cross building, I've committed `targets-rpi2` branch so you can at least run in RPi2.

```
git checkout targets-rpi2
make -f targets/rpi2/Makefile.rpi2
```
output binary should exist as `targets/rpi2/bin/jerryscript` file. I think it can be merged after #733 lands.

For performance testing, hope someone will add some comments how to.

jiangzidong, 2016-01-06:
@seanshpark Thanks
1,we can remind users to `apt-get install gcc-arm-linux-gnueabihf` and `apt-get install g++-arm-linux-gnueabihf`first
2, should add `.PHONY all jerry clean` in your makefile, otherwise, it may tell "all is already up-to-date" if user has file `jerry` in the main folder

seanshpark, 2016-01-06:
@jiangzidong , thanks for the comments, applied to above comment and to git branch. is it ok?

jiangzidong, 2016-01-06:
@seanshpark Yes, that works.Thanks!
1,And if we set `LTO = ON`, then copy the final jerryscript into the RPI2.
2, then use the tools/run-perf-test.sh
the perf data is similar with those posted by @egavrin

seanshpark, 2016-01-06:
@jiangzidong , ok, good. I haven't tested with `LTO=ON` but seems working :)

egavrin, 2016-01-13:
LTO=ON is set by default.

ruben-ayrapetyan, 2016-01-14:
@jiangzidong, currently we measure performance on Rpi2 with the following steps:
 - build engine through `make TOOLCHAIN=build/configs/toolchain_linux_armv7l-hf.cmake release.linux -j` on a host
 - run measurement on board through `setarch linux32 -R ./tools/run-perf-test.sh ./jerry.1 ./jerry.2 5 500 sunspider-1.0.2`

closed 2016-01-20 by jiangzidong