jerryscript-project/jerryscript #797: Some questions about snapshot

GrowingM, 2016-01-05:
By [#661](https://github.com/Samsung/jerryscript/issues/661), I know some details about Jerry-snapshot. But I still don't know how to use the snapshot mode. The flowing is my way. My `test.js` is:
```javaScript
var a = 1;
var b = 2;
var c;
c = a + b;
print(c);
```
I run `./debug.linux --dump-snapshot-for-global test_snapshot test.js `. Then, run `./debug.linux --exec-snapshot test_snapshot `. There is a trouble:
```
ICE: Assertion 'snapshot_offset < snapshot_size' failed at /home/ubun/Desktop/jerryscript/jerry-core/parser/js/bc/bytecode-data.cpp(bc_load_bytecode_data):759.
Error: ERR_FAILED_INTERNAL_ASSERTION
```
When I modified the code  `'snapshot_offset < snapshot_size' `of file `/jerry-core/parser/js/bc/bytecode-data.cpp`: `759   JERRY_ASSERT (snapshot_offset <= snapshot_size);`
It can work. This is a bug, right? The snapshot `test_snapshot` is:
```
0200 0000 0000 0000 2000 0000 0100 0000
e96c 7ae0 0000 0000 0400 0000 0101 0000
0061 0101 0000 0062 0101 0000 0063 02ce
0000 0000 0000 0000 5800 0000 1c00 0000
1800 0000 0000 0000 0300 0000 0000 0000
3801 0000 1200 0101 1201 0102 2902 0001
0083 0301 3902 02ff 06ff ffff 0400 0000
0a00 0000 1000 0000 0400 0000 0400 0000
0400 0000 0a00 0000 1000 0000 1600 0000
```
Howerver, when I show the byte-code of the file `./debug.linux --show-opcodes test.js `. The result is:
```
// var a = 1;
// var a = 1;
// 
// 
// 
// 
// 
// var a = 1;
  0:         reg_var_decl  253  253  253     // 253 tmp regs, 253 local variable regs, 253 argument variable regs
  1:           assignment    0    1    1     // a = 1: SMALLINT;
// 
  2:           assignment    1    1    2     // b = 2: SMALLINT;
// 
// 
  3:             addition    2    0    1     // c = a + b;
// 
  4:               call_n  131    3  253     // 
  5:                 meta    2    2  255     // 
  4:               call_n  131    3    1     //  // REWRITE
// 
  0:         reg_var_decl    1    0    0     // 1 tmp regs, 0 local variable regs, 0 argument variable regs // REWRITE
  6:                  ret                    // ret;
LITERALS:
0x96e744 [  8] a : STRING
0x96e74c [  8] b : STRING
0x96e754 [  8] c : STRING
0x96e75c [  4] print : MAGIC STRING [id=206] 
0x96e760 [ 32]  : EMPTY RECORD
  0:         reg_var_decl    1    0    0     // 1 tmp regs, 0 local variable regs, 0 argument variable regs
  1:           assignment    0    1    1     // a = 1: SMALLINT;
  2:           assignment    1    1    2     // b = 2: SMALLINT;
  3:             addition    2    0    1     // c = a + b;
  4:               call_n  131    3    1     // 
  5:                 meta    2    2  255     // tmp131 = print (c);
  6:                  ret                    // ret;
3
```
So my second question is what is the relationship of bytecode and snapshot? My bytecode file can run directly with jerryscript?
Thanks a lot.

wateret, 2016-01-06:
Answering second question, basically snapshot is pre-compiled bytecode. Basically Jerryscript parses `.js` file and generates bytecode at runtime before executing the code. With snapshot you can separate parse and execution. Comparison with Java, Snapshot is just like `.class` file. 

AFAIK snapshot is mainly for reducing memory consumption of IoT.js. STM32F4Discovery supports XIP so we can save RAM consumption as much as bytecode size of native modules.

GrowingM, 2016-01-06:
@wateret Thanks. I have got more information about bytecode from [Internals](http://samsung.github.io/jerryscript/internals/). It is very nice for constrained devices to run snapshot code. I have to learn more about execution of snapshot.

ruben-ayrapetyan, 2016-01-12:
@ValyrianSteel, besides byte-code, snapshot also contains snapshot header and literals. Because of this, raw dump of byte-code is not the same as raw dump of snapshot.

However, in current version, snapshot includes byte-code as is. For your example, the byte-code is the following (7 instructions - each is 4-byte length):
```
3801 0000 1200 0101 1201 0102 2902 0001
0083 0301 3902 02ff 06ff ffff
```

ValyrianSteel, 2016-01-14:
@ruben-ayrapetyan Thank you very much. I see, the snapshot  contains snapshot header, literals and bytecode header and bytecode.

ruben-ayrapetyan, 2016-01-14:
@ValyrianSteel, you are welcome.

closed 2016-01-14 by ruben-ayrapetyan