jerryscript-project/jerryscript #802: Integrity checks in snapshot load routines

ruben-ayrapetyan, 2016-01-13:
Hello.

We need to add integrity checks to snapshot loader to ensure that an incorrectly composed snapshot wouldn't lead to memory corruption or other issues during the load procedure.

egavrin, 2016-01-13:
It should verify checksum and engine's version at least.

ruben-ayrapetyan, 2016-01-13:
There is snapshot version field in current version of snapshot header that is checked upon load.

egavrin, 2016-01-13:
So, what about adding checksum?)

ruben-ayrapetyan, 2016-01-13:
I think that this is good idea, but the checksum comparison wouldn't catch errors in case snapshot is intentionally malformed.

egavrin, 2016-01-13:
Such issues can be checked only with appropriate testing, i.e snapshot can successfully pass internal test-suite and test262.

ruben-ayrapetyan, 2016-01-14:
@egavrin, could you, please, clarify what kind of issues do you mean?

ruben-ayrapetyan, 2016-01-14:
There was an offline discussion on the issue and we've concluded that this is issue is about integrity checks that should guarantee that only snapshot images that could be successfully read in conformity to the current snapshot format, should be considered correct by snapshot loader, and all other snapshot images should lead to correct indication of inconformity by the loader, without any memory corruptions or produce of incorrect run-time structures etc.

LaszloLango, 2016-02-09:
@egavrin, @ruben-ayrapetyan, @zherczeg, can we close this issue?

zherczeg, 2016-02-12:
I think we could think about this later for untrusted snapshots, but currently all snapshots are trusted. I close this for now, but it can be reopened later.

closed 2016-02-12 by zherczeg