jerryscript-project/jerryscript #821: Copyright of logo

Seo-Young, 2016-01-25:
Thankfully, I would like to announce our project logo with @davidsz 's help.

Before announce it, I would like to gather opinions of community members on copyright.
We have two options to choice.
1) Szeged University --> @davidsz create this logo and he belongs to Szeged.
2) JerryScript Community --> The community members have decided this so it's reasonable, too. 

Please share to me your opinions.

seanshpark, 2016-01-25:
+1 for 2)

Seo-Young, 2016-01-26:
+1 for 2)

davidsz, 2016-01-26:
2) is okay for me. :)

akiss77, 2016-01-26:
I'm not a lawyer, but my best understanding is: Simply the voting on the logo allows no claim on the copyright. On the other hand, @davidsz has the right to *transfer* the copyright to the community. (He did not create that piece of art in his capacity as an employee of the University of Szeged, so he is free to do so.) Fortunately, he seems to be OK with such a transfer.
(Sorry if the above seems to be too picky. But it's better to be on clear terms if it comes to legal things. And once again, I'm no lawyer, so it might be worth checking with someone who has a degree in this.)

Seo-Young, 2016-01-27:
Thanks @akiss77 for making clear it. 
Then let's put "JerryScript Community" on the copyright and announce the logo.
@lemmaa  please upload the logo with the copyright under MIT.

closed 2016-01-27 by Seo-Young

lemmaa, 2016-02-01:
@Seo-Young  PR https://github.com/Samsung/jerryscript/pull/831.