jerryscript-project/jerryscript #827: Build Jerryscript for Contiki OS/Galileo board

ssultana, 2016-01-28:
Hi,

I want to build JerryScript for Contiki OS (C runtime) and Intel Galileo board. Can you please point me where to start from, which makefiles, compile/linker flags, etc. to change? Would it be easier to change in the iot.js build system?

Thanks.

seanshpark, 2016-01-28:
Hello, @ssultana, 
`targets` folder contains more simple Makefile about porting to other target boards or embed os.
I suggest to start from reading [targets/esp8266/Makefile.esp8266](https://github.com/Samsung/jerryscript/blob/master/targets/esp8266/Makefile.esp8266) file. To see what environment variables are meaning, take your time reading [Makefile](https://github.com/Samsung/jerryscript/blob/master/Makefile)
Create a folder in `targets` and start from adding, say, `Makefile.gallileo`, and add one by one from `Makefile.esp8266`. Start with adding a source file to add an entry function to be called by the os and to call JerryScript APIs, like [jerry_run.cpp](https://github.com/Samsung/jerryscript/blob/master/targets/esp8266/source/jerry_run.cpp) in `esp8266`.
Leave any questions if you need help.

ssultana, 2016-01-31:
Hi,

Thanks for the pointer. I am trying to build Jerryscript as a library and link it to the Contiki application. It, however, shows me linker errors like 'undefined references to setjmp', 'undefined references to stderr'. I see a 'setjmp' implementation in 'jerryscript/jerry-libc/arch/x86-32.h' and 'jerryscript/jerry-libc/target/linux/jerry-asm.S'. Also, there is 'stderr' definition in 'jerry-libc/include/stdio.h'. Why does it show linker errors?

seanshpark, 2016-02-01:
Hello @ssultana , `setjmp/longjmp` is system function and should be provided by the OS(or the system). `setjmp` inside JerryScript is used for MCU, bare-bone mode (only JerryScript is flashed to the embedded hardware without any OS). As for ESP8266 you can read it about in bottom of https://github.com/Samsung/jerryscript/blob/master/targets/esp8266/docs/ESP-PATCHFORJERRYSCRIPT.md document. 
`stderr` is in the standard C library, that should be provided by the system or the compiler. File `jerry-libc/include/stdio.h` is for MCU as explained above. 
I don't know about Contiki but if there is any kind of `libc` in Contiki, then you should use that one not the one provided in the JerryScript or you may get some execution problems.

ssultana, 2016-02-01:
Thanks @seanshpark . Contiki OS uses 'newlib' library and it doesn't have setjmp/longjmp implementation. I read the ESP8266 documentation but could not understand why it would need the setjmp library. Thanks for your clarification. As far as I understand, I have to implement setjmp/longjmp for Contiki and x86-32 (for galileo board). Can you please point me to some setjmp/longjmp implementation for x86-32? I see setjmp/longjmp implementation in Nuttx (iotjs source code) but that is for ARM.

akiss77, 2016-02-01:
@ssultana, I'm not familiar with either Contiki or the Galileo board, but I've seen newlib before. According to my best knowledge, it is a fully fledged libc implementation - which should not require the writing of your own setjmp/longjmp implementation.

With just a quick googling, I've found the following:
* https://github.com/contiki-os/contiki/blob/master/platform/galileo/bsp/libc/build_newlib.sh Here it seems that Contiki just grabs to official newlib sources and builds them with some patches applied and some options disabled. But nothing hints at setjmp/longjmp being left out from the library.
* https://sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=tree;f=newlib/libc/machine/i386;h=e50baa84dc83cd3fc331e3adf01dc4723e2b741b;hb=refs/heads/master Here it seems that newlib does contain setjmp/longjmp implementation for i386.

I hope the above helps.

ssultana, 2016-02-01:
Thanks @akiss77. Now I think the issue is not about setjmp/longjmp implementation, rather linking to the proper libc. Actually, I downloaded iotjs source code and use './tools/build.py --buildtype=release --target-arch=x86 --buildlib' command to generate the libjerrycore.a library. I then use this library for my contiki application. What I see is, building this way includes "/usr/include/setjmp.h" instead of 'newlib' implementation of setjmp, and so on for other library functions. In the iotjs/tools/build.py, I see a snippet like: 
if option.target_os == 'linux':
        cmake_opt.append('-DUSE_COMPILER_DEFAULT_LIBC=YES')

Should I change it as follows to provide the Contiki provided libraries?
if option.target_os == 'contiki':
        cmake_opt.append('-DEXTERNAL_LIBC_INTERFACE=' +
                         join_path([option.contiki_home, 'include']))

Thanks.

seanshpark, 2016-02-01:
Hi @ssultana ,
```
if option.target_os == 'contiki':
    cmake_opt.append('-DEXTERNAL_LIBC_INTERFACE=' + join_path([option.contiki_home, 'include']))
```
would do. I think you are using IoT.js to drop JerryScript library. One thing you should know is that current IoT.js doesn't use latest version of JerryScript.
BTW, have you changed also `-DCMAKE_TOOLCHAIN_FILE` for Contiki/Galileo ?

ssultana, 2016-02-02:
Thanks @seanshpark , I didn't think about changing `-DCMAKE_TOOLCHAIN_FILE`. Galileo board has x86 processor, Contiki has gcc compiler. Should my x86-contiki.cmake be like:

`include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME EXTERNAL)
set(CMAKE_SYSTEM_PROCESSOR i686)

set(EXTERNAL_CMAKE_C_COMPILER gcc)
CMAKE_FORCE_C_COMPILER(${EXTERNAL_CMAKE_C_COMPILER} GNU)` 

Yes, I am using IoT.js build system. I found it has --buildlib option and thought it might be easier to build libjerrycore.a this way. However, I downloaded IoT.js as a zip file and put latest jerryscript source in the iotjs-master/deps/jerry/ folder. I hope it doesn't cause any problem. Then, I only changed the `-DEXTERNAL_LIBC_INTERFACE` in the iotjs-master/tools/build.py. After I change `-DCMAKE_TOOLCHAIN_FILE`, do I have to make any more changes?

Thanks.

seanshpark, 2016-02-02:
Hi @ssultana , 
If you are using IoT.js as building JerryScript for Contiki you may have to deal also with `--target-os` option and search `nuttx`, handle as such with the results. IoT.js target for non-linux is `nuttx` and it can give you hints what to do for Contiki.

For `x86-contiki.cmake` you wrote is ok but if using with `-DEXTERNAL_LIBC_INTERFACE` may have to change lots of things. Please look inside CMakeFiles.txt in JerryScript.

If your aim is to get only JerryScript library I suggest to look inside [Makefile.mbedk64f](https://github.com/Samsung/jerryscript/blob/master/targets/mbedk64f/Makefile.mbedk64f) file (its simpler than `esp8266`) and make modifications appropriate for contiki-x86. I think dealing only with
```
jerry:
	mkdir -p $(INTERM)
	mkdir -p $(OUTPUT)
	mkdir -p $(COPYTARGET)
	cmake -B$(INTERM) -H./ \
	 -DENABLE_LTO=OFF \
	 -DENABLE_VALGRIND=OFF \
	 -DCMAKE_TOOLCHAIN_FILE=build/configs/toolchain_external.cmake \
	 -DUSE_COMPILER_DEFAULT_LIBC=YES \
	 -DEXTERNAL_CMAKE_SYSTEM_PROCESSOR=arm7-m \
	 -DEXTERNAL_CMAKE_C_COMPILER=arm-none-eabi-gcc \
	 -DEXTERNAL_CMAKE_CXX_COMPILER=arm-none-eabi-g++ \
	 -DEXTERNAL_COMPILE_FLAGS="$(EXT_CFLAGS)" \
	 -DEXTERNAL_MEM_HEAP_SIZE_KB=$(JERRYHEAP)

	make -C $(INTERM) $(TYPE).external
	cp `cat $(INTERM)/$(TYPE).external/list` $(OUTPUT)/.
	cp $(OUTPUT)/lib$(TYPE).jerry-core.a $(COPYTARGET)/libjerrycore.a
	cp $(OUTPUT)/lib$(TYPE).jerry-fdlibm.third_party.lib.a $(COPYTARGET)/libfdlibm.a
``` 
may build JerryScript libraries for you, and remove the rest sections. You may have to change `JERRYHEAP`, `EXT_CFLAGS`, `EXTERNAL_CMAKE_SYSTEM_PROCESSOR`, `EXTERNAL_CMAKE_C_COMPILER`, `EXTERNAL_CMAKE_CXX_COMPILER` and `COPYTARGET`

Open https://github.com/Samsung/jerryscript/tree/master/targets/mbedk64f page (the content of readme.md file) about building.

> nother question: Is Jerryscript multi-threaded?

No, it's single-threaded.

ssultana, 2016-02-02:
To make sure that I understand, you suggest to use 'Jerryscript' source and build system (instead of IoT.js) to build `libjerrycore.a.` In this regard, I should:

** Make a directory `jerryscript/targets/contiki-x86.` I think the only source file to add in `contiki-x86/source` is `jerry_port.cpp`. I use `libjerrycore.a` as a library to the Contiki application and call `jerry_simple_run()` from Contiki app. So, I don't need to implement `js_enty()`, `js_exit()`, etc. as in `mbedk64f/jerry_run.cpp`, nor anything like `mbedk64f/main.cpp`. 

** Write a `jerryscript/targets/contiki-x86/Makefile.contikix86` and set the variables you mentioned. 
```
COPYTARGET=targets/contiki-x86/libjerry
EXT_CFLAGS= i have to figure out
JERRYHEAP= i have to figure out
... ...

jerry:
... ...
-DCMAKE_TOOLCHAIN_FILE=build/configs/toolchain_external.cmake \
-DEXTERNAL_LIBC_INTERFACE=contiki_home\include \
-DEXTERNAL_CMAKE_SYSTEM_PROCESSOR=x86 \
-DEXTERNAL_CMAKE_C_COMPILER=i686-linux-gnu-gcc or only gcc is enough? \
-DEXTERNAL_CMAKE_CXX_COMPILER=i686-linux-gnu-g++ or only g++ is enough?
... ...

```
** To build libjerrycore.a, I just need to run: `make -f targets/contiki-x86/Makefile.contikix86 jerry`

No changes needed to `jerryscript/CMakeLists.txt` or any other file. Do I need a makefile like `targets/mbedk64f/source/makejerry.cmake`? I don't see where it's getting called.

One concern: Jerryscript is built using g++ (as I set `EXTERNAL_CMAKE_CXX_COMPILER` to g++) but Contiki has no g++ compiler. Would there be any kind of section alignment issue, etc. to use `libjerrycore.a` as a library in Contiki?

Thanks.

seanshpark, 2016-02-02:
Hi @ssultana ,

> I think the only source file to add in contiki-x86/source is jerry_port.cpp

Yes, it's ok. `mbedk64f` codes are one of an example.

> -DEXTERNAL_CMAKE_C_COMPILER=i686-linux-gnu-gcc or only gcc is fine?

I don't know about Contiki/Galileo but I think c++ compiler you are using should work. 

> Why would I need cross-compiling?

I also don't think it's a cross-compiling as it's x86 core. But you are compiling in linux host and for Contiki so it may be something like `cross-platform-compiling with same CPU architecture`. One way is to use `toolchain_external.cmake` file with required options, `-DEXTERNAL_LIBC_INTERFACE=contiki_home\include` so that it will compile for/with Contiki system libraries. Another one would be adding new one, something like `toolchain_contiki_i686.cmake` with correct values. I think you can try with the first one and go on with the second one.

> Also, how to determine the value of JERRYHEAP?

In JerryScript heap is allocated as static variable, default and maximum is set to 256KB. As for Galileo board, I see that it has 256MB of memory and I assume you are not using all of them so you can remove the `-DEXTERNAL_MEM_HEAP_SIZE_KB=$(JERRYHEAP)` line and let heap size be 256KB as default.

> Any changes needed to jerryscript/CMakeLists.txt or any other file?

No, there is no need.

> Do I need a makefile like targets/mbedk64f/source/makejerry.cmake ?

Also, no need to. Makefile.contikix86 will run cmake with `jerryscript/CMakeFiles.txt`

>  don't see where it's getting called.

You mean `cmake -B$(INTERM) -H./ \` line ?

> Would there be any kind of section alignment issue, etc. to use libjerrycore.a as a library in Contiki?

We had some memory alignment issues and all known are fixed. I'm not sure nothing will go wrong but let's fix them when it occurs for Contiki/Galileo. 

One more thing, JerryScript also needs `libfdlibm.a` so please link with this file.

ssultana, 2016-02-03:
I successfully ran a Contiki application, invoking jerry_simple_run(), on Galileo board. Thanks a lot @seanshpark . You were very responsive and helpful the whole time :)

seanshpark, 2016-02-03:
@ssultana , you're welcome and glad to hear its work!

closed 2016-02-03 by ssultana

denzelwamburu, 2017-02-24:
@ssultana Can you share the process or the code. Am really interested on this one.