jerryscript-project/jerryscript #842: Nominating Zsolt Borbely (bzsolt) for Committer status

zherczeg, 2016-02-05:
Hi,

I would like to nominate Zsolt Borbely to becoma a committer. His non-trivial patches are:

#78 Implemented Array.prototype.slice().
#86 Implement Array.prototype.splice().
#99 Implement Object.prototype.propertyIsEnumerable().
#138 Implement String.prototype.slice()
#163 Implement Function.prototype.apply()
#196 Fix the indexing of Array builtin functions.
#321 Implement Function.prototype.bind function
#386 Implement String.prototype.localeCompare function
#530 Implement String.prototype.split function
#729 Add option to enable all-in-one build mode

Any seconds?

galpeter, 2016-02-05:
+1 (also afaik he also worked on the CBC parts)

Seo-Young, 2016-02-18:
If there isn't any other opinions, let's go for next(put his name on committer list).

zherczeg, 2016-02-18:
Added to: https://github.com/Samsung/jerryscript/wiki/Assigned-people

closed 2016-02-20 by akiss77