jerryscript-project/jerryscript #853: Build [error: conversion to ‘uint8_t {aka unsigned char}’ from ‘int’ may alter its value]

CodeKittey, 2016-02-08:
Some Errors from make.log:
**OS:** Ubuntu 15.04, x86_64
**Build**: make (build_all_native)
**Revision**: 91a0514

```
jerry-core/parser/js/js-parser.cpp: In function ‘uint8_t* parser_encode_literal(uint8_t*, uint16_t, uint16_t)’:
/jerryscript/jerry-core/parser/js/js-parser.cpp:411:16: error: conversion to ‘uint8_t {aka unsigned char}’ from ‘int’ may alter its value [-Werror=conversion]
       *dst_p++ = (uint8_t) (literal_index >> 8) | CBC_HIGHEST_BIT_MASK;
```
```
In function ‘void lexer_construct_regexp_object(parser_context_t*, int)’:
/jerryscript/jerry-core/parser/js/js-lexer.cpp:1751:19: error: conversion to ‘uint16_t {aka short unsigned int}’ from ‘int’ may alter its value [-Werror=conversion]
     current_flags |= (int16_t) flag;
```

LaszloLango, 2016-02-09:
@CodeKittey, what GCC version do you use?

LaszloLango, 2016-02-09:
@CodeKittey, it should work now. Could you check the current master?

CodeKittey, 2016-02-09:
Sry couldnt figure out during the day :(

Used gcc version 5.2.1
targets debug.linux and release.linux works fine 

with build_all_native i get now another error:

```
jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp: In function ‘ecma_builtin_global_object_encode_uri_helper’:
jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp:1066:19: error: ‘MEM[(lit_utf8_byte_t *)&octets + 3B]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   lit_utf8_byte_t octets[LIT_UTF8_MAX_BYTES_IN_CODE_POINT];
                   ^
```

LaszloLango, 2016-02-10:
Related issue: #657

closed 2016-02-12 by LaszloLango