jerryscript-project/jerryscript #878: Cmake does not stop on error "gcc: error: qc: No such file or directory"

zoobab, 2016-02-12:
I am trying this version  bc826540297c90ccc39b70f10ed3392d4aefb7e6, and the "make" of cmake does not stop on error:

`
zoobab@chuchu /home/zoobab/soft/jerryscript/build2 [162]$ make -j5
[  0%] Linking C static library libdebug.jerry-libc.linux.lib.a
[  0%] Building C object third-party/fdlibm/CMakeFiles/debug-cp.jerry-fdlibm.third_party.lib.dir/k_tan.c.o
[  0%] Building C object third-party/fdlibm/CMakeFiles/debug.jerry-fdlibm.third_party.lib.dir/k_tan.c.o
gcc: error: qc: No such file or directory
gcc: error: libdebug.jerry-libc.linux.lib.a: No such file or directory
jerry-libc/CMakeFiles/debug.jerry-libc.linux.lib.dir/build.make:190: recipe for target 'jerry-libc/libdebug.jerry-libc.linux.lib.a' failed
make[2]: *** [jerry-libc/libdebug.jerry-libc.linux.lib.a] Error 1
CMakeFiles/Makefile2:2350: recipe for target 'jerry-libc/CMakeFiles/debug.jerry-libc.linux.lib.dir/all' failed
make[1]: *** [jerry-libc/CMakeFiles/debug.jerry-libc.linux.lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  0%] Building C object third-party/fdlibm/CMakeFiles/debug-cp.jerry-fdlibm.third_party.lib.dir/e_pow.c.o
[  0%] Building CXX object jerry-core/CMakeFiles/debug.jerry-core.dir/vm/vm.cpp.o
[  0%] Building CXX object jerry-core/CMakeFiles/debug-cp.jerry-core.dir/mem/mem-heap.cpp.o
`

Is there an option in CMakeLists.txt to make it stop?

akiss77, 2016-02-12:
The first issue here is (which cannot be really solved, only worked around, see below) that the build is parallel: `make` is started with `-jN` to execute independent recipes in separate processes. And if one of the children dies on error, `make` does not kill the rest but waits for them to terminate properly (or they can also error). From the GNU make manual (5.4 Parallel Execution):

"If make terminates for any reason (including a signal) with child processes running, it waits for them to finish before actually exiting."

This is why you see extra lines appearing after the firs error message (it is even printed that `make[1]: *** Waiting for unfinished jobs....`). If it's really important for you to get it stop then you might try `-j1` (but that will slow down the build process).

As for the *real* error: the build system most probably does not find the `ar` tool. (At least that's when I see `cq` or `qc` in error messages.)

At this line:
https://github.com/Samsung/jerryscript/blob/master/CMakeLists.txt#L111
cmake tries to infer the path to `ar` from the path to the compiler by replacing "gcc" with "gcc-ar" -- and this can go wrong if your system has `ar` smoewhere else or your compiler is not named `gcc`.

What platform are you on (arch, OS, distro) and what compiler tool chain do you use?

zoobab, 2016-02-15:
I am on sabayon-linux:

zoobab@chuchu /home/zoobab/soft/jerryscript [26]$ cat /etc/sabayon-release 
Sabayon Linux amd64 15.11
zoobab@chuchu /home/zoobab/soft/jerryscript [27]$ gcc -v
Using built-in specs.
COLLECT_GCC=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.3/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/4.9.3/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-4.9.3/work/gcc-4.9.3/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/4.9.3 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/include/g++-v4 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/4.9.3/python --enable-objc-gc --enable-languages=c,c++,java,go,objc,obj-c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo Hardened 4.9.3 p1.1, pie-0.6.2' --enable-esp --enable-libstdcxx-time --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --enable-lto --with-cloog --disable-isl-version-check --enable-libsanitizer
Thread model: posix
gcc version 4.9.3 (Gentoo Hardened 4.9.3 p1.1, pie-0.6.2) 

Gcc-ar is there:

zoobab@chuchu /home/zoobab/soft/jerryscript [29]$ which gcc-ar 
/usr/bin/gcc-ar
zoobab@chuchu /home/zoobab/soft/jerryscript [30]$ gcc-ar
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/../../../../x86_64-pc-linux-gnu/bin/ar: no operation specified

I am trying with "make release.linux -j" but I have the same error:

gcc: error: qc: No such file or directory

zoobab, 2016-02-15:
I tried under Ubuntu, it compiles fine with cmake.

However, there is no command "qc" nor "gcc-qc".

zherczeg, 2016-02-15:
Could you build with VERBOSE=1 ? And just post the command which fails. I think this qc is used when compiling fdlibm on your linux.

zoobab, 2016-02-15:
Here it is:

```
[  4%] Linking CXX static library libdebug.jerry-core.a
cd /home/zoobab/soft/jerryscript/build2/jerry-core && /usr/bin/cmake -P CMakeFiles/debug.jerry-core.dir/cmake_clean_target.cmake
cd /home/zoobab/soft/jerryscript/build2/jerry-core && /usr/bin/cmake -E cmake_link_script CMakeFiles/debug.jerry-core.dir/link.txt --verbose=1
/usr/bin/cc qc libdebug.jerry-core.a  CMakeFiles/debug.jerry-core.dir/jerry.cpp.o CMakeFiles/debug.jerry-core.dir/jerry-port.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-snapshot.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-strings.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-magic-strings.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-literal.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-char-helpers.cpp.o CMakeFiles/debug.jerry-core.dir/lit/lit-literal-storage.cpp.o CMakeFiles/debug.jerry-core.dir/rcs/rcs-allocator.cpp.o CMakeFiles/debug.jerry-core.dir/rcs/rcs-cpointer.cpp.o CMakeFiles/debug.jerry-core.dir/rcs/rcs-chunked-list.cpp.o CMakeFiles/debug.jerry-core.dir/rcs/rcs-records.cpp.o CMakeFiles/debug.jerry-core.dir/rcs/rcs-iterator.cpp.o CMakeFiles/debug.jerry-core.dir/mem/mem-heap.cpp.o CMakeFiles/debug.jerry-core.dir/mem/mem-poolman.cpp.o CMakeFiles/debug.jerry-core.dir/mem/mem-allocator.cpp.o CMakeFiles/debug.jerry-core.dir/vm/vm.cpp.o CMakeFiles/debug.jerry-core.dir/vm/opcodes.cpp.o CMakeFiles/debug.jerry-core.dir/vm/opcodes-ecma-bitwise.cpp.o CMakeFiles/debug.jerry-core.dir/vm/opcodes-ecma-relational.cpp.o CMakeFiles/debug.jerry-core.dir/vm/vm-stack.cpp.o CMakeFiles/debug.jerry-core.dir/vm/opcodes-ecma-arithmetics.cpp.o CMakeFiles/debug.jerry-core.dir/vm/opcodes-ecma-equality.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-string-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-object.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-compact-profile-error.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-referenceerror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-date-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-object-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-rangeerror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-regexp.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-date.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-math.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-type-error-thrower.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-urierror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-rangeerror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-global.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-error-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-urierror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers-json.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-array.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-regexp-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-boolean.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtins.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-typeerror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-syntaxerror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-number.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-array-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-evalerror-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-json.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-number-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-typeerror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-string.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-error.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-evalerror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-boolean-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-function-prototype.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-referenceerror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers-date.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-function.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-syntaxerror.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-lcache.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-helpers-external-pointers.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-init-finalize.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-gc.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-helpers-string.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-helpers.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-helpers-values-collection.cpp.o CMakeFiles/debug.jerry-core.dir/ecma/base/ecma-helpers-number. [... cut ...]
jerry-core/CMakeFiles/debug.jerry-core.dir/build.make:2772: recipe for target 'jerry-core/libdebug.jerry-core.a' failed
make[2]: *** [jerry-core/libdebug.jerry-core.a] Error 1
make[2]: Leaving directory '/home/zoobab/soft/jerryscript/build2'
CMakeFiles/Makefile2:3483: recipe for target 'jerry-core/CMakeFiles/debug.jerry-core.dir/all' failed
make[1]: *** [jerry-core/CMakeFiles/debug.jerry-core.dir/all] Error 2
make[1]: Leaving directory '/home/zoobab/soft/jerryscript/build2'
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
zoobab@sabayonx86-64 /home/zoobab/soft/jerryscript/build2 [17]$ 
```

akiss77, 2016-02-16:
Yep, this is the issue that's been haunting us for a while. Somehow, the build system detects your native compiler(s) at `/usr/bin/cc` and `/usr/bin/c++` (with the completed C++->C transition, this one is not relevant anymore), but also classifies it as `GNUCC`. And then the logic at https://github.com/Samsung/jerryscript/blob/master/CMakeLists.txt#L107 tries to infer the location of `ar` from that path by relacing the substring `gcc` with `gcc-ar`. But your path does not contain the `gcc` substring...

Since all that logic is guarded by a check on `ENABLE_LTO`, could you try building with `make LTO=OFF`?

kenchris, 2016-06-29:
I get this on Ubuntu as well, also with ```LTO=OFF```
```
[100%] Linking C static library librelease.jerry-core.a
ccache: error: Could not find compiler "qc" in PATH
ccache: error: Could not find compiler "qc" in PATH
jerry-libc/CMakeFiles/release.jerry-libc.linux.lib.dir/build.make:164: recipe for target 'jerry-libc/librelease.jerry-libc.linux.lib.a' failed
make[3]: *** [jerry-libc/librelease.jerry-libc.linux.lib.a] Error 1
CMakeFiles/Makefile2:1103: recipe for target 'jerry-libc/CMakeFiles/release.jerry-libc.linux.lib.dir/all' failed
make[2]: *** [jerry-libc/CMakeFiles/release.jerry-libc.linux.lib.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
jerry-libm/CMakeFiles/release.jerry-libm.lib.dir/build.make:536: recipe for target 'jerry-libm/librelease.jerry-libm.lib.a' failed
make[3]: *** [jerry-libm/librelease.jerry-libm.lib.a] Error 1
CMakeFiles/Makefile2:1817: recipe for target 'jerry-libm/CMakeFiles/release.jerry-libm.lib.dir/all' failed
make[2]: *** [jerry-libm/CMakeFiles/release.jerry-libm.lib.dir/all] Error 2
ccache: error: Could not find compiler "qc" in PATH
jerry-core/CMakeFiles/release.jerry-core.dir/build.make:2720: recipe for target 'jerry-core/librelease.jerry-core.a' failed
make[3]: *** [jerry-core/librelease.jerry-core.a] Error 1
CMakeFiles/Makefile2:2350: recipe for target 'jerry-core/CMakeFiles/release.jerry-core.dir/all' failed
make[2]: *** [jerry-core/CMakeFiles/release.jerry-core.dir/all] Error 2
CMakeFiles/Makefile2:393: recipe for target 'CMakeFiles/release.linux.dir/rule' failed
make[1]: *** [CMakeFiles/release.linux.dir/rule] Error 2
Makefile:222: recipe for target 'release.linux' failed
make: *** [release.linux] Error 2
```

zherczeg, 2016-06-30:
Hi Kenneth,

the world seems small :)

Is this happen with "make release.linux"? It seems ccache is a tool to accelerate compilation with gcc. I don't know its internals and why it depends on qc. Google also doesn't find anything useful with "ccache qc".

kenchris, 2016-06-30:
It is indeed :-) Yes this happens with make release.linux.

I tried Googling as well and didn't find anything. I will see if I can disable ccache somehow and whether that makes a difference.

zherczeg, 2016-06-30:
Here: http://stackoverflow.com/questions/2693805/how-to-use-ccache-selectively they recommend disabling cache with:

    export CCACHE_DISABLE=1

Could you try it?

galpeter, 2016-06-30:
@kenchris, @zherczeg the whole problem comes from the top level CMakeLists.txt file (around line 95):
```
   set(CMAKE_AR ${DIRECTORY_GCC}/${CMAKE_AR})
   set(CMAKE_RANLIB ${DIRECTORY_GCC}/${CMAKE_RANLIB})
```
It is guarded by the LTO flag, so one must really turn of the LTO or remove/comment out that two lines to work with ccache. 

I've also tried with ccache + LTO  on/off and for me it went fine, but I had to do a clean build.

More details:
The top level CMakeLists.txt tries to rewrite the default ar/ranlib command by searching for the configured compiler's path and replace the "gcc" part with "gcc-ar"/"gcc-ranlib" however the ccache does not provide wrappers for that thus ends up with the "ccache" for ar and ranlib. With this the command which will be used to create the archive becomes something like this: `ccache qc <archive_name> <*.o files>`

kenchris, 2016-06-30:
Thanks @galpeter !

zherczeg, 2016-07-28:
This is invalid after the new build system lands.

closed 2016-07-28 by zherczeg