jerryscript-project/jerryscript #879: [linux, compact profile] ICE: Assertion '!is_equal' failed at jerry-core/ecma/operations/ecma-lex-env.cpp(ecma_op_set_mutable_binding):219

akiss77, 2016-02-12:
When executing `tests/jerry-test-suite/13/13-011.js` with a debug compact profile build of jerry on linux, the engine crashes with status code 120 and error message as follows:

```
ICE: Assertion '!is_equal' failed at ../../../jerry-core/ecma/operations/ecma-lex-env.cpp(ecma_op_set_mutable_binding):219.
Error: ERR_FAILED_INTERNAL_ASSERTION
```
The release compact profile build does not report any error on the same input.

LaszloLango, 2016-02-15:
I checked this and the assert must be a leftover. I'll propose a PR for removing it.

akiss77, 2016-02-16:
The merge of #882 makes the original assert disappear (of course) but the same test now triggers another error:
```
[737/766] ./build/bin/debug.linux-cp/jerry  ./tests/jerry-test-suite/13/13-011.js: FAIL (120)
ICE: Assertion 'ecma_is_completion_value_empty (completion)' failed at /home/akiss/devel/jerry/jerryscript/jerry-core/ecma/operations/ecma-objects-arguments.c(ecma_op_create_arguments_object):265.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

LaszloLango, 2016-02-16:
@akiss77, let me check.

closed 2016-02-25 by LaszloLango