jerryscript-project/jerryscript #881: js snapshot execute failed on big endian machine

qdk0901, 2016-02-14:
Hi, 
  I'm poring iotjs to mips big-endian machine, but jerry_exec_snapshot is failed when execute js snapshot,  
I dig a little the code, fix endian problem in jerry_snapshot_header_t structure,  but there still OUT_OF_MEMORY exception, so I want to know if the jerryscipt can dump snapshot into big-endian mode, and how to?

zherczeg, 2016-02-15:
Currently there is no "other" endian support in Jerry, so you need a big endian machine to generate big endian snapshot. But I think this is a reasonable feature request, so I keep it in my mind. How urgent is this for you?

qdk0901, 2016-02-15:
hi,
  it not so urgent. but when I disable snapshot feature in iotjs, it still throw error 
```
root@OpenWRT:~# ./iotjs hello.js
ICE: Unreachable control path at /home/source/iotjs/deps/jerry/jerry-core/ecma/operations/ecma-regexp-object.cpp(re_match_regexp):1132 was executed.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted
```
when running a simple code on ar9331 (big endian mips soc)
```
console.log('hello')
```

zherczeg, 2016-02-15:
Do you use iot.js with a recent or with the default Jerry?

qdk0901, 2016-02-15:
I use this version
[eda775a](https://github.com/Samsung/jerryscript/commit/eda775a8ca17f34942190b9a14173730fa212c6a)
, not the default version

zherczeg, 2016-02-15:
This is a 23 days old version, and there was a lot of changes (like moving to a new byte code representation) since that. I think the latest Jerry need this patche to work with iot.js:

https://github.com/Samsung/iotjs/pull/330/files

qdk0901, 2016-02-15:
I update Jerry to the latest version and patch the iotjs, regexp problem still exist on mips big endian soc, but in mipsel soc(little endian), very thing goes right.

zherczeg, 2016-02-15:
Great to hear that! You still need a big endian machine to generate big endian snapshot, until we work on its support.

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg