jerryscript-project/jerryscript #883: Missing "make install" on linux-x64 target

zoobab, 2016-02-15:
Hi,

I am missing a "make install" in cmake or autotools to install the final binaries on the system (I am on a linux 64 bits ubuntu machine).

I presume jerryscript binaries are those:
`
release.linux
release.linux-cp
release.linux-cp_minimal
release.linux-cp_minimal-mem_stats
release.linux-mem_stats
release.linux-mem_stress_test
`
However, to be installed a linux system, those are weird binary names. What about renaming them to:
`
jerryscript
jerryscript-cp
jerryscript-cp_minimal
jerryscript-cp_minimal-mem_stats
jerryscript-mem_stats
jerryscript-mem_stress_test
`
Furthermore, there are no simple way for a newbie to understand how this JS VM works, I tried on the command line:
`
root@7448a5be284f:/mnt/jerryscript/build2# cat ../example.js 
{
  jerry_init (JERRY_FLAG_ENABLE_LOG);

  char script [] = "print ('Hello, World!');";
  jerry_parse (script, strlen (script));

  jerry_run ();

  jerry_cleanup ();
}
root@7448a5be284f:/mnt/jerryscript/build2# ./release.linux ../example.js 
root@7448a5be284f:/mnt/jerryscript/build2# 
`

Having API documentation is good, but it looks like C code, while people expect JS code (a la nodejs).

akiss77, 2016-02-15:
@zoobab Where did you get that example.js from?... It's nowhere in the project tree AFAIK. It looks like a C source indeed. If you are looking for a first JS file to execute, try tests/hello.js.

LaszloLango, 2016-07-12:
Related PR #1192

closed 2016-07-28 by zherczeg