jerryscript-project/jerryscript #911: Build error with USE_COMPILER_DEFAULT_LIBC="YES"

zoobab, 2016-02-22:
Hi,

I am trying to build the HEAD of the GIT (aa0b6215fad7ff2f5c1f88055eed622f3386b6ee) with:

```
$ mkdir build3
$ cd build3
$ cmake -DUSE_COMPILER_DEFAULT_LIBC="YES" ..
$ make
```
While with jerrylibc it compiles fine, with the GLIBC of my system (debian jessie amd64), it fails:

```
zoobab@esivisu /home/zoobab/soft/jerryscript-master/build3 [374]$ make -j1
[  0%] Building C object jerry-core/CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers-date.c.o
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c: In function ‘ecma_date_local_tza’:
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:450:19: error: storage size of ‘tz’ isn’t known
   struct timezone tz;
                   ^
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:452:3: error: implicit declaration of function ‘gettimeofday’ [-Werror=implicit-function-declaration]
   if (gettimeofday (NULL, &tz) != 0)
   ^
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:450:19: error: unused variable ‘tz’ [-Werror=unused-variable]
   struct timezone tz;
                   ^
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c: In function ‘ecma_date_daylight_saving_ta’:
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:476:19: error: storage size of ‘tz’ isn’t known
   struct timezone tz;
                   ^
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:476:19: error: unused variable ‘tz’ [-Werror=unused-variable]
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c: In function ‘ecma_date_local_tza’:
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:458:1: error: control reaches end of non-void function [-Werror=return-type]
 } /* ecma_date_local_tza */
 ^
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c: In function ‘ecma_date_daylight_saving_ta’:
/home/zoobab/soft/jerryscript-master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:484:1: error: control reaches end of non-void function [-Werror=return-type]
 } /* ecma_date_daylight_saving_ta */
 ^
cc1: some warnings being treated as errors
jerry-core/CMakeFiles/debug.jerry-core.dir/build.make:698: recipe for target 'jerry-core/CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers-date.c.o' failed
make[2]: *** [jerry-core/CMakeFiles/debug.jerry-core.dir/ecma/builtin-objects/ecma-builtin-helpers-date.c.o] Error 1
CMakeFiles/Makefile2:1827: recipe for target 'jerry-core/CMakeFiles/debug.jerry-core.dir/all' failed
make[1]: *** [jerry-core/CMakeFiles/debug.jerry-core.dir/all] Error 2
Makefile:76: recipe for target 'all' failed
make: *** [all] Error 2
```

Could you also add that "USE_COMPILER_DEFAULT_LIBC="YES"" to Travis-ci as well?

zoobab, 2016-02-22:
It builds fine if I checkout to b2f3ec225e931e5b2ddac6a249626ffce8fdf045 (17 feb, 5 days ago).

LaszloLango, 2016-02-22:
Related PR: #790 (it will be updated soon)

LaszloLango, 2016-02-23:
@zoobab, this should be fixed in 76b37f34d0df7fadbecf57ac52345c6b50631b54, please check.

zoobab, 2016-02-23:
Yes, the build is working now.

closed 2016-02-23 by zoobab