jerryscript-project/jerryscript #922: Should we keep external magic string support?

bzsolt, 2016-02-25:
Let's start a discussion about external magic string support, because there are some related functions,
but they aren't used currently. Now we can just register external magic strings, but they aren't accesssible through the API.
Is there anybody who uses, or plans to use this feature?

akiss77, 2016-02-29:
IoT.js uses external magic strings but only the `jerry_register_external_magic_strings` function (in  iotjs/src/iotjs_string_ext.cpp). It would be interesting to see how its memory consumption would change without those magic strings.

zherczeg, 2016-04-01:
I would rather keep it (or a special version of it). However, it should be automatically generated from snapshot. This would reduce the string memory consumption a lot. I would like to work on it if noone else is interested.

closed 2016-07-28 by zherczeg