jerryscript-project/jerryscript #933: How can I build jerryscript on another ARM board?

DongDongig, 2016-03-07:
I'm trying to build jerryscript on Odroid U3 or XU3 board..
As I know, someone already Did build it on RPI2....
But when i try this, I found some error 

gcc-4.8.real: error: cr: No such file or directory
gcc-4.8.real: error: libdebug.jerry-core.a: No such file or directory
make[5]: *** [jerry-core/libdebug.jerry-core.a] Error 1
make[4]: *** [jerry-core/CMakeFiles/debug.jerry-core.dir/all] Error 2
make[3]: *** [CMakeFiles/debug.linux.dir/rule] Error 2
make[2]: *** [debug.linux] Error 2

is it support to build on ARM now??

thx in advance

LaszloLango, 2016-03-07:
Related issues: #489, 878

zherczeg, 2016-07-28:
New build system should solve this.

closed 2016-07-28 by zherczeg